gnuton/asuswrt-merlin.ng #7: Adaptive QoS->Spectrum

zendilar, 2019-04-18:
Adaptive QoS->Spectrum (Traffic Manager - Spectrum)
There is not any results and it looks like empty after I pressed "Rescan" button it looks like refreshing the posses but nothing Is displayed.

gnuton, 2019-04-22:
If you run the page in chrome and press F12, do you get any error in the Javascript console?

zendilar, 2019-04-24:
1.what's the release runnning on the router?
DSL-AC68U_384.10_gnuton4_DSL_1.0.4.7.trx

2.If you browse the page QOS spectrum in chrome and you press F12, do you get any error in the Javascript console? 
in chrome I get nothing in console tab, in firefox I get the line below in console tab.
The script from “http://192.168.1.1/help_content.js” was loaded even though its MIME type (“text/html”) is not a valid JavaScript MIME type.[Learn More]

as for 3. and 4. I will come back.

juwgxibz, 2019-05-12:
> 
> 
> Adaptive QoS->Spectrum (Traffic Manager - Spectrum)
> There is not any results and it looks like empty after I pressed "Rescan" button it looks like refreshing the posses but nothing Is displayed.

The same situation.
Release:  DSL-AC68U_384.11_gnuton1_DSL_1.0.4.7.trx

gnuton, 2019-05-13:
It doesn't work because  the application which was supposed to get spectrum data was not copied from the Asus GPL. 
./src/router/spectrum/prebuild/spectrum

gnuton, 2019-05-13:
He guys could you please test this firmware? It's based on the latest release + the fix for spectrum.
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/gnuton-snapshot-feature-spectrum

juwgxibz, 2019-05-13:
It's working! Thank you!

gnuton, 2019-05-13:
excellent. It will be part of the next 384.11 release. I will close this ASA I release the fix. thanks for testing @juwgxibz !

closed 2019-05-14 by gnuton