gnuton/asuswrt-merlin.ng #10: HTTPS login page not accessable

walgarch, 2019-04-24:
Hi, and thanks for all the great work!

I'm having and issue with accessing the router management page via the HTTPS login page. I've currently set up both HTTP and HTTPS access to the router, but accessing the HTTPS page timeouts on a browser. It used to work fine on the stock firmware.

I've set it up to give me both HTTP and HTTPS access in the mean time
[image]

Please let me know if there are any logs of extra info I can provide that might help

Cheers!

gnuton, 2019-04-24:
Hi,
Since I do not have the device with me, could you please check if there is any Javascript error happening when you enable it?
Any other log would be helpful too. 
Thanks

walgarch, 2019-04-25:
Hmmm... I played around with it a bit more and found out that the HTTPS login page still works on Chrome, so this appears to be a Firefox related issue. I'll dig around for more info, but I think you can safely assume that this isn't an issue with the firmware and more likely a SSL cert issue on firefox. Thanks for taking a look! and sorry for being a pain. Feel free to close this ticket :)

closed 2019-04-25 by walgarch