gnuton/asuswrt-merlin.ng #11: Ad IPsec support for IKEv2 VPN scripts.

raidensnake, 2019-05-01:
It would be good to implement the IPsec support to allow a more secure vpn for devices that don't support OpenVPN. It would also need Strongswan as well as a change to rc. 

IKEv2 scripts: https://www.snbforums.com/threads/asus-ipsec-vpn-server.44973/page-2#post-484117

Merlin Commits to IPsec/Strongswan:

https://github.com/RMerl/asuswrt-merlin.ng/commit/d6eb2a8916aee03bca7448e66e28ebe51a7c1f9b

https://github.com/RMerl/asuswrt-merlin.ng/commit/70ed53730c33d6ad1bc406e9166a60705252c7d4

https://github.com/RMerl/asuswrt-merlin.ng/commit/fbfc8a2181b8fb0c4053fda1fb31277445e3be58

https://github.com/RMerl/asuswrt-merlin.ng/commit/db4fdc262c68dde6114c58dd127a4741b43977f3

https://github.com/RMerl/asuswrt-merlin.ng/commit/3fd2098b6cbda4b41f6852b8d25b002ff5c0da96

gnuton, 2019-05-03:
Hi,
That code is actually disabled for many routers for a good reason. Enabling it would mean to rebuild the kernel with crypto modules which would not make the kernel able to load the precompiled Asus modules.
So we can enable this option only if Asus rebuilds those new modules for us.

raidensnake, 2019-05-07:
I wonder what @Rmerl has to say as from what I can see it should work as is.

gnuton, 2019-05-07:
I have already asked. :D I may try to enable it and see what happens in a
separate branch.

raidensnake, 2019-05-07:
please do it would be interesting to see what happens.

gnuton, 2019-05-07:
Please find the image with IPSEC enabled in the release page.
I have not tested it at all, so let me know if it works.

raidensnake, 2019-05-07:
well ikev2 is responding however it's relying on ddns and a cert for it to work.

gnuton, 2019-05-07:
You should be able to re-use the cert as long as you have the same domain name. You may wanna Google or ask for assistance in the forum for this.

raidensnake, 2019-05-07:
The main problem is getting the ddns issue resolved since it says I'd need to make a ddns-start script. There needs to be a way of either ignoring the ddns or just reverse check the IP for the existing hostname.

gnuton, 2019-05-07:
good luck with the setup and let me know if you get it working

RMerl, 2019-05-07:
Review syslog.  Some kernel modules will fail to load, reporting mismatched kernel symbols.  I don't remember if the issue was with the Tuxera or the Trend Micro modules, it's been too long.

raidensnake, 2019-05-13:
I'm please to say with the help of @sinshiva and @RMerl on IRC I was able to get IKEv2 working however there were 4 issues I discovered.

1. LetsEncrypt for some reason didn't want to run so I uses a Globalsign cert.
2. Since I use a static address the post commit scripts by @odkrvs from the SMB forums had to be modified.
3. I had to change the IP range to allow lan access in the ipsec config.
4. The location of the certs in the scripts when uploaded were in the wrong place as well has having to add the private key info in ipsec.secrets and make the intermediate cert separate and upload it since it didn't import it as well as changing the location in the ipsec config file.
and 5. the config I changed is lost if I either restart or change anything.

gnuton, 2019-05-13:
So if I got it right you managed to get it running.
BTW have you seen any kernel mismatched symbols when loading the tuxera/TM modules?

raidensnake, 2019-05-13:
you mean this?

`May 13 10:06:47 kernel: tfat: disagrees about version of symbol generic_file_splice_read
May 13 10:06:47 kernel: tfat: Unknown symbol generic_file_splice_read (err -22)
May 13 10:06:47 kernel: tfat: disagrees about version of symbol generic_file_splice_write
May 13 10:06:47 kernel: tfat: Unknown symbol generic_file_splice_write (err -22)
May 13 10:06:47 kernel: tfat: disagrees about version of symbol kernel_recvmsg
May 13 10:06:47 kernel: tfat: Unknown symbol kernel_recvmsg (err -22)
May 13 10:06:47 kernel: tfat: disagrees about version of symbol unregister_sysctl_table
May 13 10:06:47 kernel: tfat: Unknown symbol unregister_sysctl_table (err -22)
May 13 10:06:47 kernel: tfat: disagrees about version of symbol register_sysctl_table
May 13 10:06:47 syslog: can't load module tfat (kernel/fs/tfat.ko): Invalid argument
May 13 10:06:47 kernel: tfat: Unknown symbol register_sysctl_table (err -22)
May 13 10:06:48 kernel: tntfs: disagrees about version of symbol generic_file_splice_read
May 13 10:06:48 kernel: tntfs: Unknown symbol generic_file_splice_read (err -22)
May 13 10:06:48 kernel: tntfs: disagrees about version of symbol generic_file_splice_write
May 13 10:06:48 kernel: tntfs: Unknown symbol generic_file_splice_write (err -22)
May 13 10:06:48 kernel: tntfs: disagrees about version of symbol kernel_recvmsg
May 13 10:06:48 kernel: tntfs: Unknown symbol kernel_recvmsg (err -22)
May 13 10:06:48 kernel: tntfs: disagrees about version of symbol unregister_sysctl_table
May 13 10:06:48 kernel: tntfs: Unknown symbol unregister_sysctl_table (err -22)
May 13 10:06:48 kernel: tntfs: disagrees about version of symbol register_sysctl_table
May 13 10:06:48 kernel: tntfs: Unknown symbol register_sysctl_table (err -22)
May 13 10:06:48 syslog: can't load module tntfs (kernel/fs/tntfs.ko): Invalid argument
May 13 10:06:48 kernel: thfsplus: disagrees about version of symbol generic_file_splice_read
May 13 10:06:48 kernel: thfsplus: Unknown symbol generic_file_splice_read (err -22)
May 13 10:06:48 syslog: can't load module thfsplus (kernel/fs/thfsplus.ko): Invalid argument`

gnuton, 2019-05-13:
thanks them. Since enabling this feature will break  Asus basic feature I would close this unless you can propose a solution for this.

closed 2019-05-13 by gnuton

raidensnake, 2019-05-13:
Not sure what features break but most work just fine.

gnuton, 2019-05-13:
The modules which cannot be loaded are the ones which allow you to mount a
USB dongle formatted in FAT/NTFS/HFS+.

raidensnake, 2019-05-14:
This is something I don't get... How can IP sec affect something that's not related to it?

gnuton, 2019-05-14:
Basically, IPSec needs some crypto kernel drivers which conflict with the
one that has been pre-built by ASUS.
So or you get IPsec or you can mount USB dongles. you cannot do both unless
we switch the the open source drivers.
This is the default config which comes from ASUS
Tuxera FAT module Support (RTCONFIG_TFAT) [Y/n] y
  Open + Tuxera FAT Support (RTCONFIG_OPENPLUS_TFAT) [N/y] n
NTFS Support (RTCONFIG_NTFS) [Y/n] y
  NTFS Support by Paragon (RTCONFIG_PARAGON_NTFS) [N/y] n
  NTFS Support by Tuxera (RTCONFIG_TUXERA_NTFS) [Y/n] y
  NTFS-3G Support (RTCONFIG_OPEN_NTFS3G) [N/y] n
HFS Support (RTCONFIG_HFS) [Y/n] y
  HFS Support by Paragon (RTCONFIG_PARAGON_HFS) [N/y] n
  HFS Support by Tuxera (RTCONFIG_TUXERA_HFS) [Y/n] y
  Kernel HFSPLUS support (RTCONFIG_KERNEL_HFSPLUS) [N/y] n

raidensnake, 2019-06-27:
Since you made changes to the newest 314.12 alpha that you stopped linking against the crypto as in the revision change here:
rc: remove unnecessary linking with libbcmcrypto

Do you think it's possible to use ikev2 now?

RMerl, 2019-06-27:
This change has nothing to do with IPSEC.  libbcmcrypto is a library used by Broadcom`s software stack, and it was unnecessarily linked to rc, which didn't use it.

raidensnake, 2019-07-22:
what I was trying to ask is now the crypto libraries are removed from the Broadcom prebuilts will ipsec work without breaking the usb support now the changes were made in 314.12?