gnuton/asuswrt-merlin.ng #12: Network map shows 0 clients in Release: 384.11-gnuton1

djspiros, 2019-05-10:
in latest Release: 384.11-gnuton1 Network Map doesnt show connected clients. even if i press view list and interface still shows 0 clients. if i press the icon above view list its scanning for devices connected but stays in scanning...
thank you!!!
[image]

raidensnake, 2019-05-11:
I can confirm this issue as well.
[image]

gnuton, 2019-05-11:
Was it working for you in the previous release (384.10-gnuton4)?
Does the number gets updated after pressing view list? It does for me in chrome and the number is right the next time.
Can you please open the page in chrome incognito. I m afraid some web page caching may be involved in this bug.

raidensnake, 2019-05-11:
I used every browser on 5 different devices. it did the same.

djspiros, 2019-05-11:
in previous release (384.10-gnuton4) worked perfect.
the problem started with 384.11-gnuton1

raidensnake, 2019-05-11:
I tested both 384.10-gnuton4 and 384.10-gnuton5 (yes it was released but was on the smb forums) and both worked just fine. Just this new one is having the issue.

gnuton, 2019-05-11:
I can reproduce it on my side too know after a factory reset and re-flash.
Thanks, guys for reporting. I will try to figure out what's the problem. Any help is welcome btw :D no idea yet what's the change that has generated this and other regressions in the web ui.

gnuton, 2019-05-14:
The problem seems to be somehow related web.c which exports the data made computed by the networkmapd daemon to the UI through the APIs http://router.asus.com/update_networkmapd.asp and http://router.asus.com/update_clients.asp. The first API re-fresh the number of client and the second one shows the details of the clients when the list is open.
384.11 contains upstream ASUS changes which touch web.c and networkmap daemon.
I'm investigating how to fix this..

gnuton, 2019-05-14:
I should have fixed this issue along with others reported here in github with this pre-release
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.11-gnuton2-alpha1
I have not tested it yet, but it would be nice to get some feedback

Keltere, 2019-05-14:
Qos and device list working now on the latest alpha release

closed 2019-05-14 by gnuton