gnuton/asuswrt-merlin.ng #15: Administration - Firmware Upgrade

zendilar, 2019-05-14:
when you go to
http://router.asus.com/Advanced_FirmwareUpgrade_Content.asp
for the seting of Scheduled check for new firmware availability yes or no option can not be saved
for the Firmware Version if you click check the page displayed is https://www.asus.com/kr/

check if it can that be fixed when you can.

gnuton, 2019-05-14:
That's a missing feature, I was planning to add it.
It requires some back-end infrastructure too to host the manifest file which contains the releases. 
It's in the TODO list

gnuton, 2019-05-20:
[image]
Ok implemented and now in testing! :D

gnuton, 2019-05-21:
tested and it will be part of the next release!

zendilar, 2019-05-21:
you are the man mate. if you want send me a link so I can test it from my side to

gnuton, 2019-05-21:
Here it is https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.11-gnuton3_alpha1

gnuton, 2019-05-26:
This feature has been released. There is a bug in the merlin script which checks for the updates which doesn't check minor firmware version by string. This has been fixed already and it's in testing in 384.11_2-gnuton2_alpha1.

gnuton, 2019-05-26:
Implemented and released the fix https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.11_2-gnuton2

closed 2019-05-26 by gnuton