gnuton/asuswrt-merlin.ng #16: DNS over TLS preset server missing

gnuton, 2019-05-14:
[image]

juwgxibz, 2019-05-14:
I can't find it in 384.11-gnuton2-alpha1

juwgxibz, 2019-05-14:
shall I switch to Italian?

gnuton, 2019-05-14:
LOL, :D the localization is quite good.. so if you can understand some
italian.. why not? be my guest.
no jokes. sorry next time I will switch to english.

gnuton, 2019-05-14:
I missed your first comment.. here it's is in English and the place where you can find it in the menus
[image]

juwgxibz, 2019-05-14:
I'm can't figure out, I'm flashed 384.11-gnuton2-alpha1, but there is not DoT in WAN in DSL WAN. The same situation I was wrote in SNB. Sorry.

juwgxibz, 2019-05-14:
What information you need if you planned to fix it?

gnuton, 2019-05-14:
The first thing I would like to know if you have the variable "dnspriv": 1
in http://router.asus.com/state.js
[image: image.png]

Keltere, 2019-05-14:
@gnuton dnspriv present for me but no DNS over TLS settings.

gnuton, 2019-05-14:
is it set to 1 or 0?
I have just published this pre-release which fixes the presets
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.11-gnuton2-alpha2 (which was the bug of this issue)

gnuton, 2019-05-14:
I will close this issue since the bug described has been fixed.
Please flash the latest pre-release (link above) and factory reset the router if you can.
Hopefully, you will see DOT in the WAN menu.
 If not please open a new issue and we will try to figure out why you cannot see it.. or why I can see it.. or I dunno :D

closed 2019-05-14 by gnuton

Keltere, 2019-05-14:
It is set to 1 on alpha2 and even on alpha1. I've installed the alpha 2 but still no DNS over TLS.

gnuton, 2019-05-15:
@Keltere I filled this issue https://github.com/gnuton/asuswrt-merlin.ng/issues/17 I think we can talk about this there.