gnuton/asuswrt-merlin.ng #17: DNS Over TLS not available for DSL

gnuton, 2019-05-14:
DNS Over TLS is available if you set your WAN to ethernet,

But not if you set to PPPoE
[image]

Keltere, 2019-05-15:
From the source code it appears the settings page is binded to `/src/router/www/Advanced_WAN_Content.asp` but the dsl page settings is `/src/router/www/sysdep/DSL-N55U/www/Advanced_DSL_Content.asp`

The commit regarding DNS over TLS settings should be this https://github.com/RMerl/asuswrt-merlin.ng/commit/6c29c5d8b36f5bf92c16fb988643756ee7e18473#diff-89bd4485b8139cf7fa5a4113b5a1a3e6

Hope is just this missing frontend code.

gnuton, 2019-05-15:
Good.. Sounds like a PR from you is coming for this feature! :D
BTW the changes you pasted are just about dnssec and in my understanding we want dnssec as well as dnspriv

Keltere, 2019-05-15:
Sorry to disappoint but i'm no expert so i would avoid any sort of PR because i still couldn't get the compiler to work and i can't test any changes.

gnuton, 2019-05-15:
who needs the compiler! fork the project enable circleci to build the code
and push your changes and you will get out the firmware images... no
worries I was joking about the PR

gnuton, 2019-05-17:
cool the task is all yours! :+1:

gnuton, 2019-05-21:
@Keltere let us know if there is any update about this task. cheers!

Keltere, 2019-05-21:
I'm trying to understand how to implement it but i think is beyond my capabilities.

gnuton, 2019-05-21:
let me know when you give up. if you need more time, no worries people can
wait :P

Keltere, 2019-05-31:
I've tried but it's really too much complex for me. Don't think i will ever be able to do something about it. Sorry.

ferongr, 2020-01-05:
I'm interested in sponsoring this bug as long as the cost is reasonable.

slovaker, 2021-09-16:
I too would like to help sponsor this particular enhancement to utilize better DNS options

gnuton, 2021-09-16:
If there is interest I will try to get this available in the next release

gnuton, 2021-09-19:
Some updates about this:
1. DNSSec and DNSPriv have been added to VDSL page
2. Stubby seems to work and connecting to the DNS servers
3. DNSSec seems to be okay
4. DNSPriv (or DNS over TLS) still doesn't work - Reason = for some reason the incoming traffic to router 53 page is not going to  stubby so the tunnel doesn't work yet.

Investigating on this.
I will add here (for me and for anyone who wanna test this in case of issues) references to commands and docs.

1. to activate stubby logs 
   nvram set stubby_debug=1
   # restart stubby somehow
   cat /tmp/stubby.log
   Other option:
   kill -9 $(echo /var/run/stubby.pid)
   /usr/sbin/stubby -C /etc/stubby/stubby.yml -l

gnuton, 2021-09-20:
I have been told that tenta.com/test TLS cannot be used to check if DoT works.
So I will validate this by checking that there is traffic on port 835. If everything is fine I will make this part of the next release.

gnuton, 2021-09-23:
Beta2 comes with a few UI fixes. I think this scan be a release candidate. As for now i do not see any other task left for this task

gnuton, 2021-09-29:
This is part of the release which is coming in a few hours.

closed 2021-09-29 by gnuton