gnuton/asuswrt-merlin.ng #18: WAN - Virtual Server / Port Forwarding

zendilar, 2019-05-15:
If you go to http://router.asus.com/Advanced_VirtualServer_Content.asp and Enable Port Forwarding and add some rules in the list, the router does not save them, after a reboot the list is empty and you have to add them again.

please take a look if that can be fixed

I am running DSL-AC68U_384.11_gnutonalpha2_DSL_1.0.4.7.trx
thanks again for the FW

juwgxibz, 2019-05-15:
It's working, but not stable.
In RMerlin's firmware the same situation, I've wrote about this bug in SMB.

zendilar, 2019-05-15:
at my home router I am running DSL-AC68U_384.10_gnuton4_DSL_1.0.4.7.trx and there is no issue with Port Forwarding rules in the list

gnuton, 2019-05-15:
Followed the steps you said  and it worked for me on 384.11_gnutonalpha2.
@zendilar please reset to factory and try again if that happens please tell me what you do to let it fail.
@juwgxibz what do you mean with it's unstable? is this issue reported in the RMerl github?

Keltere, 2019-05-15:
Tested with 384.11_gnutonalpha2 and no problem.
I use DSL connection.
@zendilar how many ports did you set? Or even better, can you give us the list?

juwgxibz, 2019-05-16:
I mean somtimes rules saved somtimes not. I don't know why.
Not reported RMerlin gintub yet, just asked question in SBM forum.

gnuton, 2019-05-16:
I will close this since I do not see an issue here. If you keep on having this prolbem we will re-open it.
cheers

closed 2019-05-16 by gnuton

jtitley, 2020-01-19:
I've had this issue with stock firmware, but it is fixed with Merlin, probably because I did a factory reset. The issue is NVRAM is full and for whatever reason, port forwarding is the first to have problems in this scenario. You can do things like clear nvram setting client_info_tmp or a factory reset (because things like multiple VPN servers can add entries in nvram and don't go away unless you clear).

My workaround was to update NAT in one of the startup scripts. Since I didn't change port forwarding often, it was fine for me to update via command line. Just nvram get vts_rulelist and then call nvram set vts_rulelist on boot.

Because I implemented this on stock, I used the post-mount hack to run this at that time.