gnuton/asuswrt-merlin.ng #19: Advanced_Wireless_Survey page not working

gnuton, 2019-05-16:
**Firmware Version Affected **
384.11-gnuton2

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
wireless survey doesn't work. spinner keeps on spinning and nothing happens
**To Reproduce**
Steps to reproduce the behavior:
1. Go to http://router.asus.com/Advanced_Wireless_Survey.asp
2. wait

**Expected behavior**
You get the list of wireless networks

gnuton, 2019-05-16:
I tried it once again and I have seen it just takes long time to scan. Eventually the list appears.

zendilar, 2019-05-23:
I followed your instructions to reproduce the problem, and nothing happens, everything seams to work fine, no disconnections and the list appears almost immediately.

I checked three times and the outcome is the same.
running: DSL-AC68U_384.11_2-gnuton1_DSL_1.0.4.7.trx

closed 2019-05-26 by gnuton