gnuton/asuswrt-merlin.ng #20: IGMP Proxy Bug

Dr-AMJ, 2019-05-19:
**Firmware Version Affected **
384.11 gnuton 2

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Thnx for your great project.
I use IPTV over DSL (NOT WAN) (multicast).. every tv channel freezing after 4.15 Minutes, I searched for this problem and I found "**igmp querier-timeout**" cause this issue, but no way to adjust it.
BTW: I used another router "TP-LINK Archer vr900" and I didn't face like this issue. IPTV work normally with tp-link router.
Can you please try to resolve this issue in next release. thanks

gnuton, 2019-05-19:
Hi,
If the bug affects all routers it must be fixed upstream. Is there any open ticket about this in the official merlin repo?

Dr-AMJ, 2019-05-20:
> Hi,
> If the bug affects all routers it must be fixed upstream. Is there any open ticket about this in the official Merlin repo?

I don't find a ticket for a similar issue... I think this issue in DSL-AC68U which connect to the internet via RJ11, RMerlin routers like RT-AC68U connect via WAN port.  so we set IPTV by LAN port, but in DSL we use Bridge to enable IPTV multicasting.

gnuton, 2019-05-20:
do you have any log to share? I'm not using the DSL of this router ATM.

gnuton, 2019-05-21:
I just merged the upstream merlin changes in this alpha.
It contains some fixed for IPTV
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.11_2-gnuton1_alpha1

Dr-AMJ, 2019-05-21:
> I just merged the upstream merlin changes in this alpha.
> It contains some fixed for IPTV
> Could you please test it out if it works for you?
> https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.11_2-gnuton1_alpha1

Thanks for your kind attention
I Updated my "DSL-AC68" with this firmware "DSL-AC68U_384.12_gnuton1_alpha1_DSL_1.0.4.7.trx"
But the problem still happens, every channel freezing after 4.15 minutes. and every channel doesn't play "Black screen" when returning to it.
System Log:
=====================================
May 21 15:10:08 acsd: scan in progress ...
May 21 15:10:08 ntp: Initial clock set
May 21 15:10:08 rc_service: ntpd_synced 795:notify_rc restart_diskmon
May 21 15:10:08 rc_service: waitting "restart_wireless" via  ...
May 21 15:10:09 acsd: scan in progress ...
May 21 15:10:09 acsd: scan in progress ...
May 21 15:10:09 acsd: scan in progress ...
May 21 15:10:09 acsd: scan in progress ...
May 21 15:10:10 acsd: scan in progress ...
May 21 15:10:10 acsd: scan in progress ...
May 21 15:10:10 acsd: scan in progress ...
May 21 15:10:10 acsd: selected channel spec: 0xe06a (100/80)
May 21 15:10:10 acsd: Adjusted channel spec: 0xe06a (100/80)
May 21 15:10:10 acsd: selected DFS-exit channel spec: 0xe06a (100/80)
May 21 15:10:10 acsd: selected channel spec: 0xe06a (100/80)
May 21 15:10:10 acsd: Adjusted channel spec: 0xe06a (100/80)
May 21 15:10:10 acsd: selected channel spec: 0xe06a (100/80)
May 21 15:10:12 disk_monitor: Finish
May 21 15:10:13 disk_monitor: be idle
May 21 15:10:21 WLCEVENTD: eth1: Assoc 20:3D:BD:F8:AE:5A
May 21 15:10:21 dnsmasq-dhcp[261]: DHCPREQUEST(br0) 192.168.1.101 20:3d:bd:f8:ae:5a 
May 21 15:10:21 dnsmasq-dhcp[261]: DHCPACK(br0) 192.168.1.101 20:3d:bd:f8:ae:5a LGwebOSTV
May 21 15:10:25 WLCEVENTD: eth1: Assoc 58:04:54:D0:11:5E
May 21 15:10:26 dnsmasq-dhcp[261]: DHCPDISCOVER(br0) 58:04:54:d0:11:5e 
May 21 15:10:26 dnsmasq-dhcp[261]: DHCPOFFER(br0) 192.168.1.110 58:04:54:d0:11:5e 
May 21 15:10:26 dnsmasq-dhcp[261]: DHCPREQUEST(br0) 192.168.1.110 58:04:54:d0:11:5e 
May 21 15:10:26 dnsmasq-dhcp[261]: DHCPACK(br0) 192.168.1.110 58:04:54:d0:11:5e android-1c93077b73407070
May 21 15:10:38 rc_service: cfg_server 836:notify_rc restart_wireless
May 21 15:10:40 kernel: br0: port 2(eth1) entering forwarding state
May 21 15:10:40 kernel: device eth1 left promiscuous mode
May 21 15:10:40 kernel: br0: port 2(eth1) entering disabled state
May 21 15:10:40 kernel: br0: port 3(eth2) entering forwarding state
May 21 15:10:40 kernel: device eth2 left promiscuous mode
May 21 15:10:40 kernel: br0: port 3(eth2) entering disabled state
May 21 15:10:42 kernel: device eth1 entered promiscuous mode
May 21 15:10:42 kernel: br0: topology change detected, propagating
May 21 15:10:42 kernel: br0: port 2(eth1) entering forwarding state
May 21 15:10:42 kernel: br0: port 2(eth1) entering forwarding state
May 21 15:10:42 kernel: device eth2 entered promiscuous mode
May 21 15:10:42 kernel: br0: topology change detected, propagating
May 21 15:10:42 kernel: br0: port 3(eth2) entering forwarding state
May 21 15:10:42 kernel: br0: port 3(eth2) entering forwarding state
May 21 15:10:43 acsd: scan in progress ...
May 21 15:10:43 acsd: scan in progress ...
May 21 15:10:44 acsd: scan in progress ...
May 21 15:10:44 acsd: scan in progress ...
May 21 15:10:44 acsd: scan in progress ...
May 21 15:10:44 acsd: scan in progress ...
May 21 15:10:45 acsd: scan in progress ...
May 21 15:10:45 acsd: scan in progress ...
May 21 15:10:45 acsd: scan in progress ...
May 21 15:10:45 acsd: scan in progress ...
May 21 15:10:46 acsd: selected channel spec: 0x1007 (7)
May 21 15:10:46 acsd: Adjusted channel spec: 0x1007 (7)
May 21 15:10:46 acsd: selected DFS-exit channel spec: 0x1007 (7)
May 21 15:10:46 acsd: selected channel spec: 0x1007 (7)
May 21 15:10:46 acsd: Adjusted channel spec: 0x1007 (7)
May 21 15:10:46 acsd: selected channel spec: 0x1007 (7)
May 21 15:10:46 acsd: scan in progress ...
May 21 15:10:46 acsd: scan in progress ...
[... 14 lines cut ...]
May 21 15:10:50 acsd: Adjusted channel spec: 0xe02a (36/80)
May 21 15:10:50 acsd: selected DFS-exit channel spec: 0xe02a (36/80)
May 21 15:10:50 acsd: selected channel spec: 0xe02a (36/80)
May 21 15:10:50 acsd: Adjusted channel spec: 0xe02a (36/80)
May 21 15:10:50 acsd: selected channel spec: 0xe02a (36/80)
May 21 15:10:50 WLCEVENTD: eth1: Assoc 20:3D:BD:F8:AE:5A
May 21 15:10:50 dnsmasq-dhcp[261]: DHCPREQUEST(br0) 192.168.1.101 20:3d:bd:f8:ae:5a 
May 21 15:10:50 dnsmasq-dhcp[261]: DHCPACK(br0) 192.168.1.101 20:3d:bd:f8:ae:5a LGwebOSTV
May 21 15:11:00 crond[269]: time disparity of 549244 minutes detected
May 21 15:11:05 WLCEVENTD: eth1: Assoc 58:04:54:D0:11:5E
May 21 15:11:05 WLCEVENTD: eth2: Assoc 88:3D:24:5B:84:64
May 21 15:11:05 dnsmasq-dhcp[261]: DHCPREQUEST(br0) 192.168.1.138 88:3d:24:5b:84:64 
May 21 15:11:05 dnsmasq-dhcp[261]: DHCPACK(br0) 192.168.1.138 88:3d:24:5b:84:64 Chromecast-Ultra
May 21 15:11:06 dnsmasq-dhcp[261]: DHCPDISCOVER(br0) 58:04:54:d0:11:5e 
May 21 15:11:06 dnsmasq-dhcp[261]: DHCPOFFER(br0) 192.168.1.110 58:04:54:d0:11:5e 
May 21 15:11:06 dnsmasq-dhcp[261]: DHCPREQUEST(br0) 192.168.1.110 58:04:54:d0:11:5e 
May 21 15:11:06 dnsmasq-dhcp[261]: DHCPACK(br0) 192.168.1.110 58:04:54:d0:11:5e android-1c93077b73407070
May 21 15:11:18 watchdog: New firmware version 384.50_gnuton4 is available.
May 21 15:17:54 syslogd started: BusyBox v1.25.1
May 21 15:17:54 kernel: klogd started: BusyBox v1.25.1 (2019-05-21 05:39:05 EDT)
May 21 15:22:05 miniupnpd[1321]: HTTP listening on port 50295
May 21 15:22:05 miniupnpd[1321]: Listening for NAT-PMP/PCP traffic on port 5351
May 21 15:28:15 syslogd started: BusyBox v1.25.1
May 21 15:28:15 kernel: klogd started: BusyBox v1.25.1 (2019-05-21 05:39:05 EDT)
May 21 15:54:34 rc_service: httpd 270:notify_rc start_webs_update

gnuton, 2019-05-21:
the logs do not really help, the IPTV code comes straight from ASUS and as for now it looks like we are waiting for asus to fix the problem if you are talking about the movistar IPTV.
I can't help you right now, but let's keep this bug issue open until it doesn't get fix.
Cheers!

gnuton, 2020-12-16:
is this bug still present in the latest v386?

gnuton, 2021-02-12:
384 release is not maintained anymore. if this issue persists on 386 please reopen the ticket.
thanks

closed 2021-02-12 by gnuton