gnuton/asuswrt-merlin.ng #24: Add Option 60 to DNS WAN configuration

Thepenguin9, 2019-11-13:
**Is your feature request related to a problem? Please describe.**
I've recently switched to Sky, and they don't support third party modems/routers. I'm forced to use my Asus router as an access point due to lack of option 60

**Describe the solution you'd like**
Add the extended fields from ethernet wan to DSL. Most importantly the option 60

**Describe alternatives you've considered**
The best thing I can think of is buying a third part modem, however this was the entire reason I bought the DSL version of the router

**Additional context**
When in the Ethernet WAN setup, it shows Option 60, but it doesn't carry over to DSL

gnuton, 2019-11-14:
Hi,
Thanks for reporting this issue with sky. Have you tried to manually change the DNSMasq config? You can force option 60 and check if that's enabled for wan interface too.

Thepenguin9, 2019-11-14:
Hey,
The DNSMasq would be able to set option 60 right?

Thepenguin9, 2019-11-14:
Hey, to add on I managed to enable and connect to the router, but editing the DNSMasq file shows nothing for me to edit

gnuton, 2019-11-17:
There will be no web interface unless that doesn't come from upstream.

closed 2019-11-17 by gnuton

Thepenguin9, 2019-11-17:
So there's no way to force it in dsl? And can you look into the option 61 not saving as well please. I can make a new issue if needs be