gnuton/asuswrt-merlin.ng #26: NVRAM usage jumped above security threshold

TulioLazarini, 2019-11-24:
**Firmware Version Affected **
384.14_beta1-gnuton2

**Is this bug present in upstream Merlin releases too?**
Not sure, as I don't have it installed elsewhere.

**Describe the bug**
After deploying this beta version, the GUI started to advise low free NVRAM space.

**To Reproduce**
Steps to reproduce the behavior:
1. Install 384.14_beta1-gnuton2
2. Log into the web interface
3. A yellow exclamation mark on the upper right corner appears to me, advising about high NVRAM usage.

**Expected behavior**
Lower NVRAM footprint.

**Screenshots**
[image]
[image]

TulioLazarini, 2019-11-24:
To be honest, I havent't seen how much free NVRAM I had when running the previous stable version. I've run `nvram show` to see the variables set there, and I haven't seen anything exceptional: half dozen of DHCP reservations, one OpenVPN profile, my config is somewhat simple...

gnuton, 2019-11-25:
Thanks a lot for reporting this issue I will be try to understand  what's behaving wrong if it's just some issue with UI or a real issue with NVRAM management.
In the meanwhile if you find something interesting to share please feel free.
Thanks!

gnuton, 2019-12-01:
Hi,
I have rolled out a new firmware pre-release featuring the latest Asus blobs and I NVRAM is back to normal for me.
Can you confirm?
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/384.14_beta2-gnuton0_alpha3

TulioLazarini, 2019-12-01:
Hi Antonio,

Thank you for your new version. I've installed it and checked on the CLI,
now we are at:

size: 60163 bytes (5373 left)

Giving the fact that modern NVRAM needs to hold hundreds of variables, I'm
afraid we're reaching the older hardware limit. There's nothing much we can
do about it, right?

Tulio Lazarini
Soli Deo Gloria

gnuton, 2019-12-01:
Hi Tulio,
Just for reference, I have reset the router and my numbers are a little bit lower than yours.
383 13 gnuton0_alpha5
size: 57688 bytes (7848 left)

383 14 beta2-gnuton0_alpha3
size: 58147 bytes (7389 left)

So there has been an increase of 450 bytes. Not great but not alarming either.
So I think I can close and mark this issue as solved.
Cheers mate for reporting and testing once again.
Feel free to re-open the ticket if required

gnuton, 2020-12-10:
This issue is back and it's due to NVRAM set to 64KB which is already small.
Asuswrt code has some flags to increase the mem size but as for now those do not work. 
Increasing the NVRAM size would need some changes in the bootlader as it happened for other routers.
To mitigate in v 386 JFFS_NVRAM is used but it seems like it still not as good as we want yet.
I reopened this task to investigate how we can improve this situation with the latest firmware

As temporary solution now this command can be used to force the router to free some nvram by removing empty vars

Useful commands:
- Liist the nvram vars which consume most of mem
  nvram show | awk '{print length(), $0 | "sort -n -r"}' | cut -d"=" -f 1 | head -n 20
-

gnuton, 2020-12-10:
Here us NVRAM JFFS in action activated in the router
gnuton@DSL-AC68U-CC88:/jffs/nvram# nvram get qos_orates
80-100,10-100,5-100,3-100,2-95,0-0,0-0,0-0,0-0,0-0
gnuton@DSL-AC68U-CC88:/jffs/nvram# cat qos_orates
80-100,10-100,5-100,3-100,2-95,0-0,0-0,0-0,0-0,0-0

gnuton, 2020-12-20:
Even using /jffs/nvram the size of nvram seems to be set to 65536 bytes.
The question now is this correct? we do have 64KB of nvram + /jffs/var which would make this a number much bigger than 64K.

[image]

gnuton, 2020-12-20:
This is the code which shows the low nvram notification
notification.js:		if((<% sysinfo("nvram.total"); %> - <% sysinfo("nvram.used"); %>) < 3000){

gnuton, 2020-12-21:
Running sysinfo cli command I do get
which doesn't match the one shown in the UI

NVRAM
size: 57628 bytes (7908 left)

gnuton, 2020-12-23:
The problems seems to be in httpd/sysinfo.c which has not been updated yet.
It fetches all variables stored by using get_all instead of using dev_get_all to skip the ones in jffs partition.

gnuton, 2020-12-23:
Fixed with commit https://github.com/gnuton/asuswrt-merlin.ng/commit/c3388d7e241868efbcc1fdf68c730580451a6971

gnuton, 2020-12-23:
it should be released with 386.00_0-gnuton0_alpha8

closed 2020-12-23 by gnuton