gnuton/asuswrt-merlin.ng #29: ppp connection does not establish automatically after reboot.

ferongr, 2020-01-05:
**Firmware Version Affected **
384.14_0

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
After rebooting the router the ppp connection doesn't establish automatically after DSL syncup. In order to connect you have to turn the Internet Connection switch off and then on again on the Network Map Internet Status panel.

**To Reproduce**
Steps to reproduce the behavior:
1. Reboot router
2. Wait for Internet Connection icon to turn yellow
3. Nothing happens.

**Expected behavior**
For the ppp connection to establish itself on its own. 

**Logs**
Log 9 minutes after reboot without doing any cation

https://gist.github.com/ferongr/e0fafb0289b1ee9d094521b1371d3e43

log continuation after toggling Internet Connection switch off and on again

https://gist.github.com/ferongr/b8998418085c44ab889d560bd144ba39

gnuton, 2020-01-06:
Hi,
From the logs I see the router connects to the DSL ppp interface correctly in both cases (boot and after switching off/on). 
You can see that from the lines "PAP authentication succeeded" and "remote IP address x.y.z.v" which are present in both logs.
As for now my guess is you got a dns or routing problem... since PPP is connected

ferongr, 2020-01-06:
I'm not knowledgeable with GNU/Linux but from comparing the logs it seems that the routing rules are not applied.

gnuton, 2020-01-09:
Would you mind to reset to factory your router and add all the configuration manually?
This looks more like a configuration problem than a firmware one.
Please let me know how it goes.

gnuton, 2020-01-15:
please re-open if necessary.

closed 2020-01-15 by gnuton