gnuton/asuswrt-merlin.ng #33: BUG Deauthenticated because sending station is leaving IBSS or ESS

Feloix94, 2020-03-11:
**Firmware Version Affected **
384.15_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**

Syslog: wlceventd_proc_event(387): eth2: Deauth_ind B4:E6:2A:DB:9A:DE, status: 0, reason: Deauthenticated because sending station is leaving (or has left) IBSS or ESS (3)

There is a bug on 5GHZ wifi
The device connected on this network resets the connection about every 3 minutes.

gnuton, 2020-03-17:
It's not a bug nor an error.
It's a log setting issue within Asus / Trendmicro code which RMerlin has no control over.
You can get more info by looking this in the forum:
https://www.snbforums.com/search/1328197/?q=wlceventd_proc_event&o=date&c[node]=42

conorriches, 2020-03-31:
Same issue here as well, I looked at https://www.snbforums.com/threads/deauthenticated-because-sending-station-is-leaving-ac68u.62451/ which indicated it might be an issue in the closed source Asus?

`Mar 31 13:02:32 syslog: wlceventd_proc_event(387): eth2: Deauth_ind DE:57:9B:85:04:D4, status: 0, reason: Deauthenticated because sending station is leaving (or has left) IBSS or ESS (3)`

gnuton, 2020-03-31:
I had a chat with other people on the forum and it looks to be just an informative message and it doesn't mean this happens because of a bug. For that reason, there is nothing to fix and  I closed this issue 15 days ago.
I re-opened it because I wanted to have a chat with @RMerlin about this, wih a more clear message to the users, but this chat has not happened yet.
As for now.. just do not get worried about this.

gnuton, 2020-04-15:
Closing since this is not an issue, but that's just an info message

closed 2020-04-15 by gnuton