gnuton/asuswrt-merlin.ng #34: Temporary WAN error stops OpenVPN clients from having internet access

conorriches, 2020-04-09:
First of all thank you for your hard work on this project. It's greatly appreciated!

**Firmware Version Affected**
384.14_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
I believe someone is experiencing a similar issue but there isn't the detail to confirm if it's the same cause. I'll link the issues, but please advice if I need to take this upstream. 
Link: https://github.com/RMerl/asuswrt-merlin.ng/issues/391

**Describe the bug**
When the WAN temporarily goes down due to an error and comes back up, any devices that are routed through an OpenVPN client do not then have internet access. It's then necessary to manually disable/enable VPN or to re-apply settings before they'll get internet access again.

**To Reproduce**
1. Set up an OpenVPN client and route at least one device through it.
2. I don't know how to reproduce the WAN error - I've tried disabling and re-enabling WAN in the GUI which works as expected, and unplugging the socket which also works as expected, but essentially the WAN needs to experience an issue it can't handle.
I've provided logs to hopefully help identify what that issue may be.

If you manually disable and re-enable the WAN the log looks like this - **this works as expected**:
 ```
Apr  8 13:45:26 WAN_Connection: WAN was restored.
Apr  8 13:45:30 rc_service: service 2765:notify_rc updateresolv
Apr  8 13:45:32 nat: apply nat rules (/tmp/nat_rules_ppp0_vlan3880)
Apr  8 13:45:32 zcip_client: configured _xxx.xxx.xxx.xxx_
Apr  8 13:45:34 openvpn-routing: Configuring policy rules for client 1
Apr  8 13:45:34 ovpn-client1[2700]: Initialization Sequence Completed
```
If however there's an issue with the WAN, the logs look like :
```
Apr  8 10:03:48 nat: apply redirect rules
Apr  8 10:03:48 WAN_Connection: Fail to connect with some issues.
Apr  8 10:03:50 WAN_Connection: WAN was restored.
Apr  8 10:03:50 nat: apply nat rules (/tmp/nat_rules_ppp0_vlan3880)
```

`Apr  8 10:03:48 nat: apply redirect rules` is the first log to show when things start going wrong. 

3. Clients that are tunneled through an OpenVPN client will now not have internet access, but do still have local access. This will remain until OpenVPN client settings are re-applied or is restarted.

**Expected behavior**
If there's an issue with the WAN, I would expect when WAN returns that the VPN tunneled clients would automatically be connected through the VPN Again.

gnuton, 2020-04-15:
Hi, @conorriches and thanks for submitting this issue.
OpenVPN component is something that comes from upstream and my builds are not touching it at all. I will keep this ticket open and linked to the upstream issue for reference.
I would also suggest you test this on a newer release.

gnuton, 2020-12-12:
Hi, would it be possible to check if this issue is still available in 386 alpha builds?

gnuton, 2021-04-13:
old issue.closing. 
Please feel free to re-open if it's still applicable to latest firmware
Thanks

closed 2021-04-13 by gnuton