gnuton/asuswrt-merlin.ng #37: Blue languages button on WebUI has incorrect encoding (shows ???)

mackonsti, 2020-04-18:
**Firmware Version Affected **
384.16_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
I do not know but was not visible on the official Asus 3.0.0.4_384_81981

**Describe the bug**
Login to WebUI, click on the Languages button (top right blue): The list that contains non-standard ASCII characters appear with a "?" instead.

**To Reproduce**
Steps to reproduce the behavior:
1. Update with 384.16_0-gnuton1
2. When ready, login to WebUI
3. Click on the languages blue button (top-right)
4. The languages using UTF-8 encoding are shown with ???

**Expected behavior**
The actual language name in the list, should be shown in UTF-8

**Screenshots**
[image]

**Notes**
Although choosing Español or Chinese, the remaining WebUI pages seem to show properly in UTF-8 encoding (presumably). But that button list is using the wrong encoding, resulting to ??? instead. I tried it on MacOS Mojave on Firefox, Safari and Chrome: all same result. The DSL-AC68U was also set to factory defaults, but nothing.

juwgxibz, 2020-04-20:
The same situation. In Merlin releases all languiges has correct characters.

gnuton, 2020-04-21:
I started to understand what's wrong here. 
Here a few notes of my findings.

That menu is generate by a C function which is defined in state.js
Here is the line which calls the the C hook
state.js:	banner_code +='<% shown_language_css(); %>';

and here is the code of the function
https://github.com/gnuton/asuswrt-merlin.ng/blob/dsl-ac68u/release/src/router/httpd/web.c#L10368

Basically the C function reads all lines a few files and output html.
My first guess is that the problem comes from a wrong locale setting in the docker image during the building process. 

The first thing I will try is to check if setting the following env var fixes the issue
LANG=en_US.UTF-8

gnuton, 2020-04-23:
Hi!
I should have fixed this issue it was indeed due to locale not set in the docker image.
The fix will be part of 384.16_0-gnuton2

[image]

closed 2020-04-24 by gnuton

mackonsti, 2020-04-24:
Thanks @gnuton !