gnuton/asuswrt-merlin.ng #38: Parent control issue

14ay, 2020-05-18:
**Firmware Version Affected **
384.17

**Is this bug present in upstream Merlin releases too?**
No rt-ac86u and rt-ac68u are fine

**Describe the bug**
Having set up Parenral Controls to turn my kids devices off at 10pm, when it turns off a device the whole dsl-rt68u turns off all connected devices. A hard reset resolves it.

Also at times it turns devices off long before their allocated time. A hard reset resolves it.

Moving to stock firmware issue go's away.

Devices have network protection on. Web and app filters also on for adult content.

gnuton, 2020-05-22:
Hi,
Thanks for reporting this issue. I haven't really got what the problem is.
If you reset your device and the problem you have mentioned go away it, it sounds you have a no valid settings stored.
Have you reset to factory AFTER flashing the merlin firmware?

14ay, 2020-05-23:
Hi Antonio,Yes I did full reset after flashing tried it twice and had the same issue. This is a brand new unit purchased last week to replace my modem and rt-ac86u. So was worried it could be a faulty unit!However it works fine on stock firmware.Just tried the new stock 3.0.0.4.386.37589 which syncs fine to vdsl2 but the dsl-ac68u will not allow throughput on download, fine on upload. So have I got a bad unit or is it firmware related?However 3.0.0.4.384_81981 Seems to work fine.Any advice plsBest regardsAlanSent from Samsung tablet.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/gnuton/asuswrt-merlin.ng/issues/38#issuecomment-632937692",
"url": "https://github.com/gnuton/asuswrt-merlin.ng/issues/38#issuecomment-632937692",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

gnuton, 2020-05-25:
Hi,
I noticed this is a common problem in Merlin, am I wrong?
https://www.snbforums.com/threads/asuswrt-merlin-384-9-parental-control-time-scheduling-broken.55046/
How do you know that this is not a problem on RT-AC68U with marlin?

14ay, 2020-05-25:
Hi Antonio,Just read that post. Think the issue stopped in Merlin V12 June 2019.I have a RT-AC86U and a RT-AC68U have no issues at all with them on Merlin V17 in fact not noticed any issue with 14.2, 15, 16 and 17 .. 14 was the last Merlin I found to have connection issues and was quickly replaced with14.2.Have tried the DSL17 again and it clean build and the issue is there. So currently using RT-AC86U with vigor 130 no issues at all with anything to be honest.Wanted to move to a dsl-ac68 as the modem connects faster on my long line with ECI cabinet.Any thoughtsThanksSent from Samsung tablet.

https://www.snbforums.com/threads/asuswrt-merlin-384-9-parental-control-time-scheduling-broken.55046/
How do you know that this is not a problem on RT-AC68U with merlin?

[
{

gnuton, 2020-05-25:
I will have a look at it and I will come back to you once I know more.

gnuton, 2020-12-16:
Hi! 
AI Protection had a big update on 386 branch. If you have the chance to try the alpha or soon the beta, please let me know if this has  been fixed.
Thanks

gnuton, 2021-02-12:
384 release is not maintained anymore. Please re-open this ticket if the bug is still present in 386 branch. thanks.

closed 2021-02-12 by gnuton