gnuton/asuswrt-merlin.ng #53: Parental Control - Scheduled time doesn't ublock blocked devices when time expires

gnuton, 2020-12-16:
**Firmware Version Affected **
386.00 alpha3

**Is this bug present in upstream Merlin releases too?**
No idea

**Describe the bug**
If you set a device to block at a certain time it will stay blocked
I know ASUS stock firmware had the exact same issue before in their earlier 386 builds, but in 386_39648 they fixed it

slesinger, 2020-12-28:
I am running stock Asus WRT and I can see the same issue there for long time. It is not resolved in latest version 3.0.0.4.386_39648-g788207a either.

davex2012, 2021-01-03:
Still no fixed for this bugs?

gnuton, 2021-01-04:
I am not sure I can fix that since PC code is not open source. I should try to find a workaround for it or we should wait for the fix for asus.

davex2012, 2021-01-10:
Is this issues is known by Asus? Where can I report?

gnuton, 2021-01-11:
I just got the newer GPLs from ASUS and I am merging them.
The new GPL is newer than what firmware that you can get from Asus.com.. so
let's see if it's fixed there.

If not I will ping asus devs and see what they say.
I will keep you posted

davex2012, 2021-01-16:
> I just got the newer GPLs from ASUS and I am merging them.
> 
> The new GPL is newer than the firmware  you can get from public Asus.com website. so let's see if it's fixed there.
> 
> 
> 
> If it's not I will ping asus devs and see what they say.
> 
> I will keep you posted
> 
> 
> 
> 
> 
> 
> 
> 
> > Is this issues is known by Asus? Where can I report?

Any news ?

gnuton, 2021-01-17:
Please test the latest alpha and let me know if it has fixed the issue. 
It's based on the latest "internal" ASUS code.

MonsterKraken, 2021-01-18:
> I am running stock Asus WRT and I can see the same issue there for long time. It is not resolved in latest version 3.0.0.4.386_39648-g788207a either.

I'm running 3.0.0.4.386_39648-g788207a, and the timing for parental controls works fine. I always set parental controls though the app however, I found the web interface unreliable

davex2012, 2021-01-18:
> Please test the latest alpha and let me know if it has fixed the issue. 
> 
> It's based on the latest "internal" ASUS code.

Where can I get that version?

gnuton, 2021-01-18:
from here:
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.00_0-gnuton0_alpha9

jtitley, 2021-03-25:
> I'm running 3.0.0.4.386_39648-g788207a, and the timing for parental controls works fine. I always set parental controls though the app however, I found the web interface unreliable

I second this. The app seems to work better. That has been my experience as well.

gnuton, 2021-04-13:
is still affecting the latest beta?

slesinger, 2021-04-17:
Unblocking on scheduled time works in latest stable version (3.0.0.4.386_41995-g272eb13 for DSL AC68U).

gnuton, 2021-04-20:
So potentially it should be fixed now, since merlin is based on GPL 42095 which is newer than the current stable stock
 41995.