gnuton/asuswrt-merlin.ng #56: Guest network can't access internet if 'Access Intranet' is disabled

Jonathan892, 2020-12-18:
**Firmware Version Affected **
386.xx

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Devices (tested on two mobiles)are denied all access if 'Access Intranet' is disabled in guest settings. The setting is designed to restrict guest accounts from having access to the internal network, which is is useful for untrusted devices, IoT etc. It was working in the first 386 beta.

**To Reproduce**
Steps to reproduce the behavior:
1. Disable 'Access Intranet' in Guest Network'
2. Attempt to connect to guest network fails

**Expected behavior**
Attempt to connect to guest network succeeds

Jonathan892, 2020-12-18:
Sorry I'm new to Github and didn't see Gnuton had already raised this

gnuton, 2020-12-19:
Do you recall in which release it was working?

Jonathan892, 2020-12-20:
Hi Antonio

It was working in DSL-AC68U_384.19_0-gnuton0_alpha3_DSL_1.0.5.1 but 
stopped working in DSL-AC68U_386.00_0-gnuton0_alpha1_DSL_1.0.5.1

Unfortunately Alpha7 has not fixed it for me.

gnuton, 2020-12-21:
When I tested this I created a new guest network and if you do so, it does work.
It doesn't work if you edit the previous guest network.

I tried to fix this by reverting the changes I though has caused the issue from alpha1.

gnuton, 2021-01-17:
thi should be fixed in the latest alpha

osmanelitok, 2021-01-18:
1. also does not work without open intranet. There is no "Sync to AiMesh Node" option in the 2nd and 3rd. It may be due to this.

gnuton, 2021-01-20:
Today I got confirmation this is an issue coming from Asus.
Upstream Merlin releases are affected by this too.
According to RMerlin there is chance this bug will be in the final release too.
I will add "waiting for asus to fix" label

gnuton, 2021-02-05:
I tested this and it seems to be solved in the beta1 for DSL-AX82U. I have not tested it on the DSL-AC68U yet.

gnuton, 2021-02-12:
Tested on DSL-AC68U running 386.01-beta1 and the bug is still there

gnuton, 2021-02-28:
I found the issue and the problem seems to be in a source file from Asus, so my guess is this bug is in all stock firmware and merlin for not HND routers. Confirmation is required about this btw.

The affected file is this:
release/src/router/rc/amas_wgn.c
In any case I managed to get the latest firmware working. The fix is still in a separate branch and it will be merged to master as I see no problems with it.

gnuton, 2021-02-28:
This is the fix for now. https://github.com/gnuton/asuswrt-merlin.ng/commit/6700436db5fe34b056637e8c0ecbe1ee57ec2349
It will be released with the next beta.

gnuton, 2021-03-01:
Released fixed with beta3

closed 2021-03-01 by gnuton