gnuton/asuswrt-merlin.ng #59: Guest Network does not assign IP address/provide internet access when connected to AiMesh node

shabooga, 2021-01-05:
**Firmware Version Affected **
384.1 (Beta 3)

**Is this bug present in upstream Merlin releases too?**
Not Sure

**Describe the bug**
Connecting to a Guest network on an AiMesh node results in "No Internet Connection" and self-assigned IP address (169.254.x.x).

**To Reproduce**
Steps to reproduce the behavior:
1. From with web ui, go to 'Guest Network'
2. Set the 'Sync to AiMesh Node' to 'All' for both primary 2.4GHz and 5GHz Guest Networks. For clarity, I used the following additional settings:
- Hide SSID: No
- Authentication Method: WPA2-Personal
- WPA Encryption: AES
- Access time: Unlimited
- Bandwidth Limiter: No
- Access Intranet: Disable
- Enable MAC Filter: Disable
3. Connect to Guest Network from any device ensuring you are connected to the AiMesh node.
4. Connection appears to be made however device reports "No Internet Connection" and uses a self-assigned IP address.
**Note:** The above only occurs when device connects to the Guest Network offered by the AiMesh node. If device connects via the main router rather than the AiMesh node, the connection is fine and an IP address is appropriately assigned.

**Expected behavior**
AiMesh node should assign a valid IP address and provide a connection to the internet.

**Screenshots**
None

gnuton, 2021-01-05:
Thanks for reporting this.
The 384 branch will be replaced soon by the newer  Merlin 386 and I do not have plans to keep the 384 alive, becasue it doesn't make sense.
Ai Mesh and guest network seems to be working right in the latest alpha according to our testers.
 If you wanna give it a try please check the latest 386 alpha. it's already available in download section.

gnuton, 2021-01-05:
Please feel free to re-open it if the 386 version is affected by the same issue.

closed 2021-01-05 by gnuton

shabooga, 2021-01-05:
Sorry for the confusion but I was using 386.1 beta 3 not 384.

shabooga, 2021-01-05:
I mistakenly input the version in this issue as 384.1 beta 3 when it should have been 386.1 beta 3. Opened new issue with correct version. Sorry for the confusion.