gnuton/asuswrt-merlin.ng #60: Guest Network does not assign IP address/provide internet access when connected to AiMesh node

shabooga, 2021-01-05:
**Firmware Version Affected**
386.1 (Beta 3)

**Is this bug present in upstream Merlin releases too?**
Not Sure

**Describe the bug**
Connecting to a Guest network on an AiMesh node results in "No Internet Connection" and self-assigned IP address (169.254.x.x).

**To Reproduce**
Steps to reproduce the behavior:
1. From with web ui, go to 'Guest Network'
2. Set the 'Sync to AiMesh Node' to 'All' for both primary 2.4GHz and 5GHz Guest Networks. For clarity, I used the following additional settings:
- Hide SSID: No
- Authentication Method: WPA2-Personal
- WPA Encryption: AES
- Access time: Unlimited
- Bandwidth Limiter: No
- Access Intranet: Disable
- Enable MAC Filter: Disable
3. Connect to Guest Network from any device ensuring you are connected to the AiMesh node.
4. Connection appears to be made however device reports "No Internet Connection" and uses a self-assigned IP address.
**Note:** The above only occurs when device connects to the Guest Network offered by the AiMesh node. If device connects via the main router rather than the AiMesh node, the connection is fine and an IP address is appropriately assigned.

**Expected behavior**
AiMesh node should assign a valid IP address and provide a connection to the internet.

**Screenshots**
None

shabooga, 2021-01-05:
Previously reported this with incorrect version but that issue was closed so opened a new one with correct version reported in bug report.

gnuton, 2021-01-05:
Hi,
The latest release I have for DSL-AC68U is alpha8.
https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.00_0-gnuton0_alpha8
What's the model of your router?

shabooga, 2021-01-05:
It’s RT-AC1900P which uses the same firmware as RT-AC68U I believe. If I install an alpha version and have issues, can I easily downgrade back to beta 3?

gnuton, 2021-01-05:
Hi!
you are posting your issues in the wrong place. Here I do supports builds only for DSL modems ad not RT.
Please use https://github.com/RMerl/asuswrt-merlin.ng

closed 2021-01-05 by gnuton

shabooga, 2021-01-05:
Oops! Apologies for that. Thanks for pointing me in the right direction.