gnuton/asuswrt-merlin.ng #63: Accelerazione NAT (Automatica) non funziona da un QoS all'altro

BoostOver, 2021-01-10:
**Firmware Version Affected **
386.00-gnuton0_alpha8

Usando il FlexQoS 1.1 da e viceversa QoS Tradizionale l'accelerazione NAT non cambia automaticamente come dovrebbe.
FlexQos equivale al QoS adattivo e supporta la funzione, invece il QoS non supporta.
Quindi dovrebbe aggionare il suo stato come ho scritto da un QoS all'altro.

gnuton, 2021-01-11:
Funziona su firmware stock?

BoostOver, 2021-01-11:
Per quanto riguarda il fw stock non saprei.......ho installato subito il tuo :-)
In opzioni di accelerazione nat (impostato in automatico) di fianco appare scritto (vado a memoria) in giallo e in inglese che l'accelerazione è attiva ,questo quando si imposta il QoS adattivo o limitazione di banda.
Quando invece si imposta il QoS tradizionale l'opzione deve disabilitare l'accelerazione e quindi deve apparire a fianco e scritto in giallo e in inglese che l'accelerazione è disabilitata........invece rimane attiva.

gnuton, 2021-01-13:
ho rilasciato una nuova pre-release alpha9. Molti dei bug dovrebbero essere
corretti.
se vuoi dargli un occhiata..

BoostOver, 2021-01-13:
Lo spettro adsl non funziona.
Che fine ha fatto fq_codel? 😭

BoostOver, 2021-01-13:
alpha9 risolve il problema

BoostOver, 2021-03-14:
A quanto pare devo aver accidentalmente chiuso la segnalazione per sbaglio.
Il problema persiste anche con l'ultima 386.02_1-gnuton0_alpha1

BoostOver, 2021-03-14:
[image]

BoostOver, 2021-03-14:
Come si evince dallo screenshot il QoS è adattivo (un attimo prima era impostato su tradizionale), il passaggio avrebbe dovuto abilitare l'accelerazione NAT ma non ha funzionato.

gnuton, 2021-03-14:
Sai se questo bug e' presente anche sui router RT?

BoostOver, 2021-03-15:
Ho googlato ma senza risultati.

gnuton, 2021-03-16:
Ho dato uno sguardo veloce a questo, e l accelerazione automatica funziona SOLO e SOLO se alcune opzioni sono disabilitate. 
Non so cosa hai abilitato per non farla funzionare visto che di cose per disabilitarla ne stanno davvero tante! :D

BoostOver, 2021-03-16:
Caso in essere:
1)Accelerazione attiva (QoS adattivo attivo),
2)Attivazione QoS tradizionale (l'accellerazione come giusto che sia viene disabilitata)
3) Attivazione QoS adattivo, l'accellerazione non viene abilitata.
4) Per abilitare l'accelerazione bisogna cliccare "applica" nella sezione di switch control della lan, in questo modo si riavvia il modem, al riavvio l'accelerazione è attiva come giusto che sia.
-- Il tutto dovrebbe funzionare in modo trasparente all'utente visto che l'opzione possibili sono, "Disabilitato" o "Automatico"--