gnuton/asuswrt-merlin.ng #64: In opzioni ADSL implementare il supporto UK come in altri vecchi modem

BoostOver, 2021-01-10:
Nelle passate edizioni c'era il supporto in "Country / ISP - Specific Setting" la possibilità di abilitare il supporto UK , qui in Italia questo permetteva di agganciare una portante più alta in upload.

gnuton, 2021-01-11:
vediamo che si puo fare... daro precedenza ai BUG e poi controllo questo. 
Se nn erro nella 386 ho visto il supporto solo per la turchia.. 
in che versione stava il supporto per l italia?

BoostOver, 2021-01-11:
Il supporto non credo che sia mai esistito per questo modello ma in altri modelli tipo il N55U D1 implementava questa ozione.

gnuton, 2021-01-14:
https://www.asus.com/me-en/Networking/DSLN55U_D1/HelpDesk_BIOS/
il DSL N55U ha un firmware e quindi driver completamene diversi nn ha GPL  e nulla che puo essere integrato qui a meno che tu sappia di come funzionava.
Il DSL-ac68U usa i country settings del DSL-AX82U. Hai visto se quello che cerchi nn e' gia li?

BoostOver, 2021-01-14:
Non ho possibilità.
Grazie lo stesso 🤗

closed 2021-01-14 by BoostOver