gnuton/asuswrt-merlin.ng #65: Bug Report ASUS

BoostOver, 2021-01-11:
**Firmware Version Affected **
386.00-gnuton0_alpha8

Appare il fumetto che avvisa di riportare bug direttamente ad ASUS linka alla pagina http://192.168.1.1/Advanced_Feedback.asp inesistente.

BoostOver, 2021-01-13:
Il problema rimane in alpha9

gnuton, 2021-01-25:
[image]

gnuton, 2021-01-25:
Dev notes:
- String is defined in ASUSGATE_note7
- code asuswrt-merlin.ng/release/src-rt/router/www/notification.js
- clicking feedback trigges the action setTimeout('document.noti_experience_Feedback.submit();'
- the form  is defined here
  state.js:	banner_code +='<form method="post" name="noti_experience_Feedback" action="/start_apply.htm" target="hidden_frame">\n';
- The code which redirect to missing(?) page is  document.noti_experience_Feedback.submit() (which is defined above).
  That function is defined as follow:  redirectFeedback:function(){location.href = 'Advanced_Feedback.asp';},

gnuton, 2021-02-07:
[image]

gnuton, 2021-02-28:
I believe this issue will go away once we switch off FRS_FEEDBACK=y flat in target.mak.
https://github.com/gnuton/asuswrt-merlin.ng/blob/DEV_master/release/src-rt/target.mak#L11
Unfortuntely the there are some linker issue that prevent us to easily switch if off and some fixes in the makefiles are needed for this to happen

gnuton, 2021-03-06:
Fixed in Beta4

closed 2021-03-06 by gnuton