gnuton/asuswrt-merlin.ng #67: DDNS with CGNAT

magistra86, 2021-01-12:
Firstly thanks for the good work.

My main problem is like the rest of our country, ISP's started using CGN (Double NAT) services. Therefore i can not use DDNS service to access my router or download master outside my home. I made hours of research and  i think there is a solution using external Method to retrieve WAN IP but that includes SSH, User scripts etc. bunch of stuff that i really dont know anything about.

İF there is a possibility can you please add that in GUI under DDNS for, a noob like me to configure and use DDNS in a CGN ISP (multiple-NAT environment)

gnuton, 2021-01-13:
Hi,
Thanks for reporting this. 
As for now I do not have plans to add this feature, but if somebody is interested I am open for contributions.

BTW since AsusWRT Merlin supports scripts, a better option and faster would be to ask somebody in the community through the  forum to help you to setup a reverse ssh proxy, ngrok or any other solution.

magistra86, 2021-01-14:
Hi, 
Thank you for your reply, can you please direct me to a forum,link etc. I will ask for their kind help.

gnuton, 2021-01-14:
http://snbforums.com/ go to wireless asus section and open a thread there. thanks.

closed 2021-01-14 by gnuton