gnuton/asuswrt-merlin.ng #68: DST not working on 386.00_0-gnuton0_alpha9

tarobun, 2021-01-12:
**Firmware Version Affected **
386.00_0-gnuton0_alpha9

**Is this bug present in upstream Merlin releases too?**
Dunno

**Describe the bug**
TimeZone is : (GMT+10) Canberra, Melbourne, Sydney
DST is set to start month 10, 1st Sunday and end month 4, 1st Sunday

System Time does not have DST applied and get a reminder that "The System time zone is different is different from your locale setting."

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Advance Settings -> Administration'
2. Scroll down to 'Basic Config'
3. Set 'TimeZone' to (GMT+10) Canberra, Melbourne, Sydney
4. Set 'DST time zone changes starts' to '10' month '1st' 'Sun' Week & days '2' hours
5. Set 'DST time zone changes ends' to '4' month '1st' 'Sun' Week & days '3' hours
4. See warning '* Reminder: The System time zone is different from your locale setting.'
5. Go to 'Advance Settings -> System Log'
6. See System Time is "* Daylight savings time is enabled in this time zone." but time does not have DST applied.

**Expected behavior**
DST applied and no warning as local time matches router time.

**Screenshots**
[image]

[image]

[image]

gnuton, 2021-02-17:
This has been fixed. The fix will be part of the next pre-release (beta2)
[image]

gnuton, 2021-02-17:
I will keep this open until this is not verified for AX82U too.

gnuton, 2021-02-21:
This has been fixed for DSL-AX82U with the 386.1_2-beta2

tarobun, 2021-03-02:
This is still not working for me in beta3.

I think this commit fixes it - https://github.com/RMerl/asuswrt-merlin.ng/pull/701

tarobun, 2021-03-09:
Works with 386.02_1-gnuton0_alpha1

gnuton, 2021-03-09:
Cool thanks for testting it out! I can close this issue then! :D

closed 2021-03-09 by gnuton