gnuton/asuswrt-merlin.ng #70: QoS tradizionale fq_codel mancante

BoostOver, 2021-01-13:
**Firmware Version Affected **
386.00-gnuton0_alpha9

**Is this bug present in upstream Merlin releases too?**
Non saprei

**Describe the bug**
Non esiste più la possibiltà di selezione fq_codel

**To Reproduce**
http://192.168.1.1/QoS_EZQoS.asp

[image]

gnuton, 2021-01-17:
e' stato rimosso upstream nella 386.1.
Vedi Changelog-NG.txt

  - REMOVED: fq_codel support for Adaptive QoS.  Due to a change
             in how Trend Micro configures QoS, it is no longer
             possible to intercept these to inject fq_codel.

closed 2021-01-17 by gnuton

BoostOver, 2021-01-17:
Il changlog fa riferimento al QoS adattivo e non a tradizionale.
Non sarò più bravo a tradurre io o chi rilascia il changelog non specifica in modo esatto.

BoostOver, 2021-01-17:
Come spiegato qui:
http://www.snbforums.com/threads/flexqos-1-1-flexible-qos-enhancement-script-for-adaptive-qos.68407/post-652510
fq_codel è ben presente e attualmente supportato in QoS adattivo da FlexQos.