gnuton/asuswrt-merlin.ng #73: The url on the "WAN" tab is error

osmanelitok, 2021-01-30:
**Firmware Version Affected **

 386.00_0-gnuton0_alpha9

**Is this bug present in upstream Merlin releases too?**

No

**Describe the bug**

The url on the "WAN" tab is "http://192.168.1.1/Advanced_VDSL_Content.asp?chg_pvc=0" while it should be "http://192.168.1.1/Advanced_WAN_Content.asp".

**Expected behavior**

There are more options when "http://192.168.1.1/Advanced_WAN_Content.asp".

**Screenshots**

[image]
[image]

gnuton, 2021-02-04:
When you press WAN on the left side bar, you should be redirected to 
1. Advanced_WAN_Content.asp if Wan type is "Ethernet Wan"
[image]
2. Advanced_VDSL_Content.asp if wan type is DSL/VDSL
[image]

3. Advanced_DSL_Content.asp if wan type is DSL/ADSL
[image]

Everything works as expected for me.

closed 2021-02-04 by gnuton

osmanelitok, 2021-02-05:
When VDSL is active, "Router Security Assessment" sends to "http://192.168.1.1/Advanced_WAN_Content.asp" for "UPnP service disabled" option. There are more features in the WAN option. If it is supposed to be this way, there is no problem.