gnuton/asuswrt-merlin.ng #74: Guest network can't access internet if 'Access Intranet' is disabled

osmanelitok, 2021-01-30:
**Firmware Version Affected **

386.00_0-gnuton0_alpha9

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**

1. also does not work without open intranet. There is no "Sync to AiMesh Node" option in the 2nd and 3rd. It may be due to this. "Bandwidth Limiter" does not work on devices other than the main modem.

When the "Sync to AiMesh Node" option is turned on in the guest network, the guest connected from aimesh devices other than the main modem connects without speed. "Bandwidth Limiter" does not work on devices other than the main modem. When the guest is given 10/1 mb speed, the main modem is also useful, but it connects at 50 / 7mb from the other aimesh device.

**Expected behavior**

Allow Guests to access the Internet when Intranet Access is disabled.

When "Sync to AiMesh Node" is active, let "Bandwidth Limiter" run.

**Screenshots**

[image]
[image]

gnuton, 2021-02-04:
Duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/56

closed 2021-02-04 by gnuton

osmanelitok, 2021-02-05:
not a copy. I've seen 56 years but there are more problems here.

gnuton, 2021-02-05:
Tested this with my AIMesh setup and everything works
Master node was a DSL-AX82U btw, but the code is the same as for the DSL-AC68U. The only difference is in some prebuilds

Internet works even if Access interent is enabled or disabled.
[image]

Bandwith is correctly limited to limits set in the configuration
[image]

osmanelitok, 2021-02-05:
yes aimesh works on main modem while connected. however, the speed limiter does not work on other modems. Also, I couldn't even connect to the network with the intranet turned off and I had to open the intranet to connect. :)

gnuton, 2021-02-06:
Can you tell me more about your setup? Are you using dsl ac68u as master
node ? What kind of modem are you using as ai mesh slave?

osmanelitok, 2021-02-06:
yes, main modem dsl-ac68u and 2 rt-ac68u connected to it.

gnuton, 2021-02-06:
Confirmed as working from @Zendilar
 https://www.snbforums.com/threads/dsl-ac68u-asuswrt-merlin-386-x-alphas-for-dsl-ac68u-are-now-available.68295/page-4#post-660642