gnuton/asuswrt-merlin.ng #75: Game and Open NAT options (port opening, QOS) none work.

osmanelitok, 2021-01-30:
**Firmware Version Affected **
384.xx

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**

Game and Open NAT options (port opening, QOS) none work. I tried it for CS GO.
When I make all the settings and say save, it does not save.

gnuton, 2021-02-06:
I have not tested this yet but works for @Zendilar
https://www.snbforums.com/threads/dsl-ac68u-asuswrt-merlin-386-x-alphas-for-dsl-ac68u-are-now-available.68295/page-4#post-660642

gnuton, 2021-02-07:
I jus tested this and I do see the new game added.
[image]
Can you please elaborate more on what are you doing?

zendilar, 2021-02-07:
Have you hit Apply, 
When you add a game or you set a port manually the gui saws it as added, but after you press apply nothing appears in the list.
Also I see that your port forwarding switch is set to disable try putting it to enable check
Zendilar

gnuton, 2021-02-17:
@osmanelitok we cannot reproduce this. I will close it.

closed 2021-02-17 by gnuton