gnuton/asuswrt-merlin.ng #76: Speed test Upload problem occurred in browsers

osmanelitok, 2021-01-30:
**Firmware Version Affected **

386.00_0-gnuton0_alpha9

**Is this bug present in upstream Merlin releases too?**

No

**Describe the bug**

- Speed test Upload problem occurred in browsers.

For example,

Edge (Chromium) Upload 1mbps
Google Chrome Upload 3 mbps
Internet Explorer Upload 6 mbps
Windows Speed Test App Upload 7 mbps

I've done these tests over and over. All the same 70mbps download is okay but there is a problem with the upload.

**Screenshots**

[image]
[image]
[image]
[image]

gnuton, 2021-01-30:
can you check this too?
 http://router.asus.com/AdaptiveQoS_InternetSpeed.asp
 The speed here will be the one measured by the router and not affected by any issue you can have in your internal network

osmanelitok, 2021-01-30:
yes there is no problem here. different results appear in each browser. This problem does not happen with software versions before this. maybe it's a problem with QOS.

[image]

gnuton, 2021-01-30:
It seems like Edge suffers of this issue on its own.
https://techcommunity.microsoft.com/t5/windows-10-servicing/ms-edge-on-chrome-basis-is-too-slow-during-the-speed-test-in-the/m-p/1546258
not sure about others browsers yet. I do use chrome and i do get same speed as in the router.

osmanelitok, 2021-01-30:
because everything works normally in alpha 8 version when there are same browsers. When I install alpha 9 the problems come back.

osmanelitok, 2021-01-30:
In addition, I have no download problems, I have a upload.

osmanelitok, 2021-02-07:
I just got the Google Chrome update and installed it. the problem is completely fixed. There is no problem at the moment. Thx.

closed 2021-02-08 by osmanelitok