gnuton/asuswrt-merlin.ng #77: Wireless settings are working but not showing this.

osmanelitok, 2021-01-30:
**Firmware Version Affected **

 386.00_0-gnuton0_alpha9

**Is this bug present in upstream Merlin releases too?**

No

**Describe the bug**

When I change the wireless network settings, all the settings are active, but when I open the page again everything stays the same.

For example; When I do Channel bandwidth 20 Mhz and Control Channel 11, the settings become active. The modem broadcasts on channel 11 and transmits it to other aimesh products.

However, when I come back to http://192.168.1.1/Advanced_Wireless_Content.asp, everything seems to be not set at all.

**Screenshots**

[image]

gnuton, 2021-02-04:
it works for me. Wondering if this is an issue related to something got messed up in the nvram of your router or what.

gnuton, 2021-02-06:
Confirmed as working for @Zendilar too
https://www.snbforums.com/threads/dsl-ac68u-asuswrt-merlin-386-x-alphas-for-dsl-ac68u-are-now-available.68295/page-4#post-660642

closed 2021-02-06 by gnuton

gnuton, 2021-02-09:
@osmanelitok have you managed to get this working in an incognito window?

osmanelitok, 2021-02-09:
> @osmanelitok have you managed to get this working in an incognito window?

yes. it works in incognito window but not in normal window.

zendilar, 2021-02-09:
I try to follow your instructions and opened a privet window on firefox  but with no success, then I opened chrome on an incognito tab but  the same thing happen, after the apply button is hit nothing stays. Or is me that I am looking at the wrong place, I suppose that the list of any game that is configured is going to be at the top or at the bottom, is that correct?

Zendilar

zendilar, 2021-02-09:
I try to follow your instructions and opened a privet window on firefox  but with no success, then I opened chrome on an incognito tab but the same thing happen, after the apply button is hit nothing stays. Or is me that I am looking at the wrong place, I suppose that the list of any game that is configured is going to be at the top or at the bottom, is that correct?

osmanelitok, 2021-02-09:
yes it worked with chrome and incognito window even I can see it from the phone. Apart from these, the page is reset as if no settings were made in all browsers.