gnuton/asuswrt-merlin.ng #78: Spettro dsl non funziona

BoostOver, 2021-02-04:
**Firmware Version Affected **
386.01-gnuton0_beta1

**Is this bug present in upstream Merlin releases too?**
Non posso testare

**Describe the bug**
Non viene mostrato lo spettro dsl

**To Reproduce**
http://192.168.1.1/Main_Spectrum_Content.asp    /Riesegui scansione

gnuton, 2021-02-04:
The problems seems to be realated to this:

http://router.asus.com/ajax_signals.asp 

returns empty arrays

bpc_us = [];
bpc_ds = [];
snr_us = [];
snr_ds = [];

gnuton, 2021-02-11:
The problem seems to be related to missing binaries

AX82U
router/dsl_autodet/
router/dsl_diag/

DSL-AC68U
router/spectrum

gnuton, 2021-02-12:
Fixed on DSL-AC68U, still broken on DSL-AX82U
[image]

BoostOver, 2021-02-12:
Grazie.

gnuton, 2021-02-12:
let's keep this open since this affect DSL-AX82U

gnuton, 2021-03-10:
I will re-open this since this is still a bug in DSL-AX82U.

gnuton, 2021-04-06:
Fix: https://github.com/gnuton/asuswrt-merlin.ng/commit/f0cadea9148ee614ce5a4cbd49b937db8f69e825

closed 2021-04-13 by gnuton