gnuton/asuswrt-merlin.ng #79: Web Access from WAN shuts down

osmanelitok, 2021-02-09:
**Firmware Version Affected **

386.01 beta 1

**Is this bug present in upstream Merlin releases too?**

No

**Describe the bug**

Although I do enable the "Enable Web Access from WAN" option, the settings are deleted every time it restarts and it becomes deactivated.

**To Reproduce**

http://192.168.1.1/Advanced_System_Content.asp  **\Enable Web Access from WAN**

**Screenshots**

[image]

gnuton, 2021-02-12:
Tested and it works for me. Cannot reproduce.

dave14305, 2021-02-12:
The Skynet firewall add-on for Merlin will disable WAN access if “Secure Mode” is enabled in the Skynet menus. Is Skynet installed? It would indicate in the syslog if it was responsible for disabling the insecure settings.

osmanelitok, 2021-02-12:
yes skynet was installed.  I removed it and it was fixed.

closed 2021-02-12 by osmanelitok