gnuton/asuswrt-merlin.ng #84: Repeater/Bridge Mode

gnuton, 2021-02-21:
**Is your feature request related to a problem? Please describe.**
RT-AC68U supports bridge mode which is not available on DSL-AC68U

Operational modes are available at http://router.asus.com/Advanced_OperationMode_Content.asp

andreabellitto, 2021-03-21:
I have implemented that, I have submitted to you a merge pull request about that:

https://github.com/gnuton/asuswrt-merlin.ng/pull/91

gnuton, 2021-03-22:
Awesome! Sure i will check that.

gnuton, 2021-04-02:
I tagged the latest code in the branch and the new images will be available in this page for download
 https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/gnuton-snapshot-feature-repeater
 
 I have tested the code and it does look to work!:D cool
 I have not tested the DSL part yet. 
 The wizard seems to have still something missing although but i would look at it because I have to complete to merge the newest GPL.. so something may be missing from there too.

andreabellitto, 2021-04-02:
As for the wizard missing things I already fixed the issue with this [commit](https://github.com/andreabellitto/asuswrt-merlin.ng-1/commit/fc0a49d00c97c7dd2d17981e50477fec37accd65)

I'm currently working on some other tweak like enabling QoS menù while in operatives mode different from DSL one, when I will end these work I will submit to you a new pull request for all this modification also

gnuton, 2021-04-07:
From now on I will be merging and roll out pre-release with this feature at each PR merge.
I will also try to keep the branch in sync with dsl-master (I do hope that doesn't cause any issue on your side).
Eventually i would love to see this feature being merged to the main line, but first I wanna be sure that DSL would work along with the other operational modes.
I have seen you are switched the QIS_Wizard from 55U to BLUECAVE, and I do still need to test it.

gnuton, 2021-04-10:
I have noticed the new dsl-ac68u wizard page  looks a little bit like the one used byDSL-AX82U. 
[image]
But the background is white and maybe the picture of the modem is not visible or the file is not present.
This is a reminder for checking that http://router.asus.com/images/WANunplug.png file exists.

andreabellitto, 2021-04-17:
Just FYI, in the last two weeks I used my DSL-AC68U in Mesh node operation mode attached to the main router (a new ASUS router I just bought) ZenWiFi AX XT8 and it works beautifully, a lot better than in repeater mode, speed are 4 times higher and wi-fi stability is awesome, a great improvement from simple repeater mode.

gnuton, 2021-04-20:
Yes AI Mesh is great, but that works only among ASUS routers.

BoostOver, 2022-08-16:
News?