gnuton/asuswrt-merlin.ng #88: Openvpn does not work after upgrade to 386.01

geoffroyguich, 2021-03-08:
**Firmware Version Affected **
386.01

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
I have two OpneVPN client connection. They both do not work after upgrade to 386.01.
[image]

**To Reproduce**
I reseted to factory setting and created the openvpn connection but still does not work.

gnuton, 2021-03-08:
can you please share the logs?

geoffroyguich, 2021-03-08:
Here are 2 log files.

One of the 384 firmware with a successful connection and one with the 386.

Thank you

Kind regards
Geoffroy

[Geoffroy Guichard - Chat @ Spike](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=xni7n)	[xni7n]

May  5 05:05:09 syslogd started: BusyBox v1.25.1
May  5 05:05:09 kernel: klogd started: BusyBox v1.25.1 (2020-05-02 02:41:58 EDT)
May  5 05:05:09 kernel: Linux version 2.6.36.4brcmarm (root@a2c8f8f439b9) (gcc version 4.5.3 (Buildroot 2012.02) ) #1 SMP PREEMPT Sat May 2 02:47:43 EDT 2020
May  5 05:05:09 kernel: CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=10c53c7f
May  5 05:05:09 kernel: CPU: VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
May  5 05:05:09 kernel: Machine: Northstar Prototype
May  5 05:05:09 kernel: Ignoring unrecognised tag 0x00000000
May  5 05:05:09 kernel: bootconsole [earlycon0] enabled
May  5 05:05:09 kernel: Memory policy: ECC disabled, Data cache writealloc
May  5 05:05:09 kernel: MPCORE found at 19020000
May  5 05:05:09 kernel: PERCPU: Embedded 7 pages/cpu @c8215000 s5472 r8192 d15008 u65536
May  5 05:05:09 kernel: pcpu-alloc: s5472 r8192 d15008 u65536 alloc=16*4096
May  5 05:05:09 kernel: pcpu-alloc: [0] 0 [0] 1 
May  5 05:05:09 kernel: Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 60416
May  5 05:05:09 kernel: Kernel command line: root=/dev/mtdblock2 console=ttyS0,115200 init=/sbin/preinit earlyprintk debug
May  5 05:05:09 kernel: PID hash table entries: 1024 (order: 0, 4096 bytes)
May  5 05:05:09 kernel: Dentry cache hash table entries: 32768 (order: 5, 131072 bytes)
May  5 05:05:09 kernel: Inode-cache hash table entries: 16384 (order: 4, 65536 bytes)
May  5 05:05:09 kernel: Memory: 128MB 128MB = 256MB total
May  5 05:05:09 kernel: Memory: 255464k/255464k available, 6680k reserved, 0K highmem
May  5 05:05:09 kernel: Virtual kernel memory layout:
May  5 05:05:09 kernel:     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May  5 05:05:09 kernel:     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
May  5 05:05:09 kernel:     DMA     : 0xf7e00000 - 0xffe00000   ( 128 MB)
May  5 05:05:09 kernel:     vmalloc : 0xd0800000 - 0xf0000000   ( 504 MB)
May  5 05:05:09 kernel:     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
May  5 05:05:09 kernel:     modules : 0xbf000000 - 0xc0000000   (  16 MB)
May  5 05:05:09 kernel:       .init : 0xc0008000 - 0xc003d000   ( 212 kB)
May  5 05:05:09 kernel:       .text : 0xc003d000 - 0xc03af000   (3528 kB)
May  5 05:05:09 kernel:       .data : 0xc03c8000 - 0xc03eb2e0   ( 141 kB)
May  5 05:05:09 kernel: SLUB: Genslabs=11, HWalign=32, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
May  5 05:05:09 kernel: Hierarchical RCU implementation.
May  5 05:05:09 kernel: 	RCU-based detection of stalled CPUs is disabled.
May  5 05:05:09 kernel: 	Verbose stalled-CPUs detection is disabled.
May  5 05:05:09 kernel: NR_IRQS:256
May  5 05:05:09 kernel: MPCORE GIC init
May  5 05:05:09 kernel: External imprecise Data abort at addr=0x0, fsr=0x1c06 ignored.
May  5 05:05:09 kernel: MPCORE Global Timer Clock 400000000Hz
May  5 05:05:09 kernel: Calibrating delay loop... 1595.80 BogoMIPS (lpj=7979008)
May  5 05:05:09 kernel: pid_max: default: 32768 minimum: 301
May  5 05:05:09 kernel: Mount-cache hash table entries: 512
May  5 05:05:09 kernel: CPU: Testing write buffer coherency: ok
May  5 05:05:09 kernel: MPCORE Private timer setup CPU0
May  5 05:05:09 kernel: Calibrating local timer... 399.853MHz.
May  5 05:05:09 kernel: L310: cache controller enabled 16 ways, CACHE_ID 0x410000c8, AUX_CTRL 0x7a130000
May  5 05:05:09 kernel: CPU1: Booted secondary processor
May  5 05:05:09 kernel: MPCORE Private timer setup CPU1
May  5 05:05:09 kernel: Brought up 2 CPUs
May  5 05:05:09 kernel: SMP: Total of 2 processors activated (3191.60 BogoMIPS).
May  5 05:05:09 kernel: devtmpfs: initialized
May  5 05:05:09 kernel: atomic64 test passed
May  5 05:05:09 kernel: NET: Registered protocol family 16
May  5 05:05:09 kernel: Found a Unknown NAND flash:
May  5 05:05:09 kernel: Total size:  128MB
May  5 05:05:09 kernel: Block size:  128KB
May  5 05:05:09 kernel: Page Size:   2048B
May  5 05:05:09 kernel: OOB Size:    64B
May  5 05:05:09 kernel: Sector size: 512B
May  5 05:05:09 kernel: Spare size:  16B
[... 518 lines cut ...]
Mar  8 21:44:54 ovpn-client1[2444]: UDP link local: (not bound)
Mar  8 21:44:54 ovpn-client1[2444]: UDP link remote: [AF_INET]86.194.123.126:12346
Mar  8 21:44:54 ovpn-client1[2444]: TLS: Initial packet from [AF_INET]86.194.123.126:12346, sid=9eee7f8c 45f50c10
Mar  8 21:44:54 ovpn-client1[2444]: VERIFY OK: depth=1, CN=ChangeMe
Mar  8 21:44:54 ovpn-client1[2444]: VERIFY KU OK
Mar  8 21:44:54 ovpn-client1[2444]: Validating certificate extended key usage
Mar  8 21:44:54 ovpn-client1[2444]: ++ Certificate has EKU (str) TLS Web Server Authentication, expects TLS Web Server Authentication
Mar  8 21:44:54 ovpn-client1[2444]: VERIFY EKU OK
Mar  8 21:44:54 ovpn-client1[2444]: VERIFY X509NAME ERROR: CN=raspberrypi_026c2337-661a-4eec-a702-7a9012cb41fe, must be raspberrypi_026c2337-661a-4eec-
Mar  8 21:44:54 ovpn-client1[2444]: OpenSSL: error:1416F086:lib(20):func(367):reason(134)
Mar  8 21:44:54 ovpn-client1[2444]: TLS_ERROR: BIO read tls_read_plaintext error
Mar  8 21:44:54 ovpn-client1[2444]: TLS Error: TLS object -> incoming plaintext read error
Mar  8 21:44:54 ovpn-client1[2444]: TLS Error: TLS handshake failed
Mar  8 21:44:54 ovpn-client1[2444]: SIGUSR1[soft,tls-error] received, process restarting
[... cut ...]
Mar  8 21:44:59 ovpn-client1[2444]: Incoming Control Channel Encryption: Using 256 bit message hash 'SHA256' for HMAC authentication
Mar  8 21:44:59 ovpn-client1[2444]: TCP/UDP: Preserving recently used remote address: [AF_INET]86.194.123.126:12346
Mar  8 21:44:59 ovpn-client1[2444]: Socket Buffers: R=[122880->122880] S=[122880->122880]
Mar  8 21:44:59 ovpn-client1[2444]: UDP link local: (not bound)
Mar  8 21:44:59 ovpn-client1[2444]: UDP link remote: [AF_INET]86.194.123.126:12346
Mar  8 21:44:59 ovpn-client1[2444]: TLS: Initial packet from [AF_INET]86.194.123.126:12346, sid=b7be62db 7f2a0545

gnuton, 2021-03-09:
Hi,
the following lines:
 Mar 8 21:44:43 ovpn-client1[2444]: OpenSSL:
error:1416F086:lib(20):func(367):reason(134)
 Mar 8 21:44:43 ovpn-client1[2444]: TLS_ERROR: BIO read tls_read_plaintext
error
 Mar 8 21:44:43 ovpn-client1[2444]: TLS Error: TLS object -> incoming
plaintext read error
 Mar 8 21:44:43 ovpn-client1[2444]: TLS Error: TLS handshake failed

from the lines it looks like there is something not right in the
certificate. Is it old or somehow corrupted?

maybe
openssl verify -CAfile <ca.crt>  cert1.pem cert2.pem .....
or
openssl x509 -CAfile <ca.crt> -dates -noout -in cert1.pem

could help to check that

geoffroyguich, 2021-03-09:
The certificate is less than 1 year old, I understand for openvpn they are valid 10 years.
The certificate works on the 384 ...
I did factory reset the 68U, erased the JFF, and reload the ovpn file. Still the exact same issue.
I saw on snbforum a similar case with some using the merlin firmware, and like me trying to connect openvpn to a raspberry PI. Maybe the problem is on merlin?
Best
Geoffroy

[Geoffroy Guichard - Chat @ Spike](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=xo9vi)	[xo9vi]

geoffroyguich, 2021-03-09:
I solved the problem by removing the Verify Server Certificate Name

Kind regards

[Geoffroy Guichard - Chat @ Spike](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=xoxix)	[xoxix]

On March 9, 2021 at 7:44 GMT, Geoffroy Guichard <geoffroy@yahoo.com> wrote:

The certificate is less than 1 year old, I understand for openvpn they are valid 10 years.
The certificate works on the 384 ...
I did factory reset the 68U, erased the JFF, and reload the ovpn file. Still the exact same issue.
I saw on snbforum a similar case with some using the merlin firmware, and like me trying to connect openvpn to a raspberry PI. Maybe the problem is on merlin?
Best
Geoffroy

[Geoffroy Guichard - Chat @ Spike](https://spikenow.com/r/a/?ref=spike-organic-signature&_ts=xo9vi)	[xo9vi]