gnuton/asuswrt-merlin.ng #89: Not possible to upgrade the firmware on my other DSL-AC68U devices remotely

zendilar, 2021-03-18:
**Firmware Version Affected **
Stable: 386.01_2-gnuton1

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
The remote device responds when I upload the firmware but it stacks to (applying settings) and the update never starts.

**To Reproduce**
Remotely connect to the device, go to administration, go to firmware update, upload the firmware from local computer. 
**Expected behavior**
The remote device responds when I upload the firmware  (applying settings) and after that the update starts and the bar with the procedure percentage % appears  

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2021-03-18:
I will check that. Thanks for reporting. 
BTW are you sure you do not have any firewall in between?

zendilar, 2021-03-18:
> 
> 
> I will check that. Thanks for reporting.
> BTW are you sure you do not have any firewall in between?

nothing changed from my end in the firewall settings from the last stable 384.17_0-gnuton1
I will completely disable the firewall to both my end and the remote device and check it out.

gnuton, 2021-03-18:
Can you also test this running the uploading in incognito mode in chrome?
If you use chrome and you open the chrome dev console by pressing F12 (on
linux or windows, not sure on mac)
you should be able to see if there are errors in the Javascript or in the
communication with the router too.

gnuton, 2021-04-17:
Bug confirmed: From what I see he wan connection gets reset when the upload starts.
For this reason if you upload something from remote you get disconnected and you the upload doesn't complete.

gnuton, 2021-04-22:
I tested this and it works with 386.02_2-gnuton0_beta1

zendilar, 2021-04-22:
I will check it as soon as posible

zendilar, 2021-05-04:
unfortunately it doesn't work the device i try to remotely upgrade the firmware has already DSL-AC68U_386.02_2-gnuton1_DSL_1.0.5.3 installed.
I followed all the above procedure but the result is the same.
please advise on the steps you took so I can repeat the procedure.

gnuton, 2021-05-04:
Hi,
I have been upgrading remotely all latest firmwares from the last 2 weeks
at least.
So i tried this more than 20 times and it always succeed. But my guess is
that if the connection is not fast enough maybe something can happen at
runtime.. and this is what you are facing. When it seems to be stuck.. it s
uploading the firmware...

What is your uploading speed from where you are trying to upload the
firmware?

zendilar, 2021-05-05:
Thank you for the quick response, the first thing that came in my mind was exactly that, but that is not the case because all of my lines are 10 Mbps upload 100 Mbps download, but because I had the same assumption I left the browser window open for a long time more than 10min just to be sure. 
Up until 384.17_0-gnuton1 one of my lines was a very bad adsl 24Mpbs that it was actually 1 Mbps up 10 Mbps down and there was no problem to remote update even with that speed.

I am sorry for busting your @@ with that issue but it is a problem for me that I have 4 devices to have to go onsite to update them.

If there is something else that is changed and it needs any configuration let me know thank you again for the grate firmwares.

gnuton, 2021-05-05:
I will re-open this. for further investigations.
Are you sure you do not have any script or anything which may affect this?

zendilar, 2021-05-05:
mate if you have gnuton you never go back to stock so I don't know : )

gnuton, 2021-05-05:
LOL. okay.. I actually do back to stock to double check if I did something
wrong or the problem comes from upstream (asus).
Let me know you get any other detail as workaround I guess it should be
possible to flash it from ssh.

I guess you can use this as workaround for now.
mtd-write -i /path/to/file.trx -d linux && reboot

from:
https://www.snbforums.com/threads/how-can-force-flash-fw-using-telnet-or-ssh.11558/#post-72160