gnuton/asuswrt-merlin.ng #93: Dual WAN - DSL/WAN not working

MHKalantarian, 2021-03-27:
**Firmware Version Affected **
386.01_2gnuton1

**Is this bug present in upstream Merlin releases too?**
Don't know

**Describe the bug**
I tried enabling dual wan to switch between dsl/wan using failover mode but when one of them is down it won't switch and I'm using ping to google.com which is timed out when I check in the router too. There is an script in snbforums for better switching maybe implement that? 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Dual WAN
2. Enable Dual WAN
3. Select failover mode and check allow failback
4. Set primary wan as WAN and secondary wan as DSL
5. Timer settings are every 5 second, 12 timeout to switch to dsl and 4 timeouts to switch back to wan
6. Set ping method to google.com

**Expected behavior**
One of the connections is always on hot standby and it should switch when the other one fails the conditions

gnuton, 2021-04-02:
Hi!
Does it work if you use DSL/ethernet LAN?

MHKalantarian, 2021-04-02:
DSL as primary? Gonna test it and report the results

MHKalantarian, 2021-04-02:
> Hi!
> Does it work if you use DSL/ethernet LAN?

Yes failover/failback works perfectly except one little strange thing that I dont know how it happened. Im pinging google on my CMD and ping works when failing over but after it failed back to primary (DSL) ping stopped responding for some reason

gnuton, 2021-04-05:
Nice to hear that worked out.
I think I have to remove WAN from the list because there is no WAN ethernet port on DSL routers, and that can be misleading.
I am not sure yet if it's mapped to any of the ethernet port either as for now.

1. are you pinging google by DNS or by IP? You could try to ping 8.8.8.8 and see if it works
Does the ping works again when you force your machine to re-connect to the router? 
Does the ping to google or any other extnal service work if you ping it from the router itself (http://router.asus.com/Main_Analysis_Content.asp)

MHKalantarian, 2021-04-09:
> Nice to hear that worked out.
> I think I have to remove WAN from the list because there is no WAN ethernet port on DSL routers, and that can be misleading.
> I am not sure yet if it's mapped to any of [...]

All the lan ports on the router can be converted to wan ports, I am currently using the Lan port #4 as wan to connect to my 4G router.

gnuton, 2021-04-09:
how do you rename the ETH to WAN? am I missing something?

DSL-AX82U has WAN, ETH1, ETH2, ETH3 and ETH4.
DSL-AC68U doesn't have WAN port. but has ETH1-4 that can be used as WAN,
but there is no "WAN" port.
Original firmware and the router mode doesn't have WAN port DSL-AC68U.
I do hope we are on the same page.

MHKalantarian, 2021-04-09:
> how do you rename the ETH to WAN? am I missing something? DSL-AX82U has WAN, ETH1, ETH2, ETH3 and ETH4. DSL-AC68U doesn't have WAN port. but has ETH1-4 that can be used as WAN, but there is no "WAN" [...]

All the ports have WAN support, there is no dedicated WAN port but you can config all the ETH ports to bridge the connection passing through them and broadcast it

gnuton, 2021-04-13:
ok sorry I got confused because I had something corrupted on my router and I was also seeing the WAN port beside the eth1-4.
I cleaned my router and I do see there is no WAN port so good.. only ethX wans now on DSL-AC68U.
Sorry for the confusion.

Back to your issue:
pinging google doesn't work if you go form PRIM ->SEC -> PRIMARY WAN.

if ping works between P -> S it should work also when switch back to S->P... but I cannot gie you an aswer if I do not see more logs.
you may wanna check if this is a problem of DNS or routing, can you ping google from router itself? when switching back?

MHKalantarian, 2021-04-14:
Everything works fine when DSL is on primary and WAN is on secondary the problem is with the reverse. Logs doesnt tell much about switching but if you think it helps i can post the log

gnuton, 2021-04-15:
Thanks for the update I will edit the title. 
Since there is a good workaround I will let asus fix this and mark this as low priority.

gnuton, 2021-06-28:
closed as won't fix

closed 2021-06-28 by gnuton