gnuton/asuswrt-merlin.ng #96: Web Admin Locks Up After Firmware Upgrade to 386.01_2-gnuton1

jtitley, 2021-04-08:
When I did a firmware upgrade from 384, everything worked, but I could not get the admin website. It froze. I was able to SSH in and tried to restart the httpd, but it didn't work.

What I did notice was that there was a Merlin reference to a database upgrade, be patient. But I waited what I thought was long enough.

Inspection revealed that a task was attempting to do something to /jffs/.sys/nc/nt_center.db. It looked like it was rebuilding, but got stuck in a loop (I could see the journal files being created and then it starts over again).

I just deleted the file, and then all was well.

I'm assuming this is the Asus Notification Center, so who cares.

Don't really expect a fix, this is just more of an FYI to anyone who has a similar issue, that was the fix that worked for me.

gnuton, 2021-04-10:
Thanks for sharing this.

I would recommend to anyone upgrading from 384 to perform:
- format /jffs
- reset to factory the old firmware
- flash the new firmware
- reset to factoryt the new one to be sure that default values are correctly applied and replace the old ones.

gnuton, 2021-04-10:
since there is nothing I can do for this, I would close this bug issue.

closed 2021-04-10 by gnuton