gnuton/asuswrt-merlin.ng #97: Unable to set VLAN tag on Ethernet WAN / Impossibile impostare il  VLAN tag impostando la WAN tramite Ethernet

ander089, 2021-04-09:
**Firmware Version Affected **
386.01_2-gnoton1

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
When I try to enable the 802.1Q, a javascript error raises.
After that, when i try to set a number into VLAD ID field between the defiden range (and also into 802.1P field), after clicking "Save" nothing changes ad no error was raised. The enable radio button returns unselected and VLAN ID and 802.1P fields are empty. 

**To Reproduce**
Steps to reproduce the behavior:
1. Into section WAN > Dual WAN select Ethernet WAN like Primary WAN and Save
2. Into section WAN > Internet Connection set the 802.1Q to enabled (javascript error raises)
3. In the same section, fill VLAN ID and 802.1q fields and click Save
4. After reload occurs, these fields are empty

**Expected behavior**
When I save and the page was reloaded, I expect that data are saved and from ifconfig I will see a new interface called vlan<VLAN ID>

**Screenshots**
[image]

gnuton, 2021-04-10:
Thanks for reporting. I noticed this and I will tak a look. Grazie

gnuton, 2021-04-24:
DSL-AX82U is not affected by this since it runs a different code.
DSL-AC68U change_dsl_dhcp_enable function is not defined in the scope at all and honestly it doesn't make any sense.
In any case the problem is not related to that function. 
Tested on original firmware and it works there. At moment my best guest is that the problem is in the C funciton which applies the settings.

gnuton, 2021-04-25:
Hi! 
I do have analized the code and this is the report.

In a nutshel:
The temporary workaround is to set these vars from ssh

ewan_dot1q // enables 0-1
ewan_vid // VLAN ID 1-4094
ewan_dot1p // 802.1P 0-7

The problem:
I should enable RTCONFIG_FRS_FEEDBACK flag back, or move out the vars from being dependent on FRS_FEEDBACK.

The Analysis:

1. when apply button is pressed a form with current data is sendto start_apply.htm
   ewan_dot1q=1&ewan_vid=300&ewan_dot1p=2 and group_id=&modified=0&action_mode=apply&action_script=restart_wan_if%3Brestart_stubby&action_wait=5
2. start_apply.htm in the end calls the hook update_variables();
3. validate_apply
   my quick understanding is that this function loops all the nvram variables in router_defaults array.
   this array is defined in shared/defaults.c
4. start_apply.htm returns no_changes_and_no_committing(); most probably because the vars above are not in the array

More about the problem:
The problem is related to these variables enabled only if RTCONFIG_FRS_FEEDBACK is enabled, but this has been disabled in merlin. Most probably it won't affect upstream because this is just related to DSL
 
        { "ewan_dot1q", "0", CKN_STR1, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },
	{ "ewan_vid", "0", CKN_STR4, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },
	{ "ewan_dot1p", "0", CKN_STR4, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },
#endif /* RTCONFIG_DSL */
#endif /* RTCONFIG_FRS_FEEDBACK */

1. when apply button is pressed a form with current data is sendto start_apply.htm
   ewan_dot1q=1&ewan_vid=300&ewan_dot1p=2 and group_id=&modified=0&action_mode=apply&action_script=restart_wan_if%3Brestart_stubby&action_wait=5
2. start_apply.htm in the end calls the hook update_variables();
3. validate_apply
   my quick understanding is that this function loops all the nvram variables in router_defaults array.
   this array is defined in shared/defaults.c
4. start_apply.htm returns no_changes_and_no_committing(); most probably because the vars are not in the array

The problem is related to these variables enabled only if RTCONFIG_FRS_FEEDBACK is enabled, but this has been disabled in merlin
 
        { "ewan_dot1q", "0", CKN_STR1, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },
	{ "ewan_vid", "0", CKN_STR4, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },
	{ "ewan_dot1p", "0", CKN_STR4, CKN_TYPE_DEFAULT, CKN_ACC_LEVEL_DEFAULT, CKN_ENC_DEFAULT, 0 },
#endif /* RTCONFIG_DSL */
#endif /* RTCONFIG_FRS_FEEDBACK */

gnuton, 2021-04-25:
Fixed. 
The fix will be available with the new beta2

closed 2021-04-25 by gnuton