gnuton/asuswrt-merlin.ng #100: ASUS DSL AX82U PROBLEM

grup2008, 2021-04-29:
ASUS DSL AX82U PRAOBLEM

hello vdsl i am using 100 mbit internet

Although I did the stability setting to 6 db, the snr setting of 8.3 db remains locked

My defined speed is 102398 but I can't pull 6 db because the stability setting is not working.

Speed stays at 92500 as SNR 8.3 db is stuck

Stability settings are not working in vdsl I ask you to fix the problem

gnuton, 2021-04-29:
does it work with the stock firmware?

grup2008, 2021-04-30:
dns filter does not work on the latest software 386.02_2

https://imgur.com/VCw5865

gnuton, 2021-04-30:
Please open different ticket for differet issues. DNS filtering works on DSL-AC68U. Not tested on DSL-AX82U yet... but the code should be the same one. I will give it a try.

gnuton, 2021-09-06:
dns filter has been fixed and the other issue has been already reported in another issue

closed 2021-09-06 by gnuton