gnuton/asuswrt-merlin.ng #102: SINCE LAST UPDATE 2.4Ghz old 20Mhz died

gaizkitapowers, 2021-05-07:
**Firmware Version Affected **
384.02_4

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
wifi 2.4Ghz 20Mhz died.

Steps to reproduce the behavior:
1. Go to 'WIFI SETUP - GENERAL TAB - BANDWITH'
2. Click on 'EITHER 20Mhz or 20/40Mhz'
3. Scroll down to '....'
4. See error

**WIFI 2.4 Ghz hides and refuses to run**

**Screenshots**
Can´t because it is a missing thing

gnuton, 2021-05-07:
Hi,
If you have upgraded from the previous stable, there were no changes in WiFi modules at all
Please see the release notes https://github.com/gnuton/asuswrt-merlin.ng/releases/tag/386.02_4-gnuton1
Wifi 2.4 should work as always, I guess this may be related to something went wrong during the flashing or bad settings or some settings which are not compatible with WiFi adapters in your network.

Could you please reset to factory and test it again?

gaizkitapowers, 2021-05-07:
Hi. I’m honoured, your highness! I understand and feel ashamed if so.

I’ve downgraded already and it works fine on 386.2.  I even thought that could be any interference with new fans attached the very same day (now I laugh – LOL).

Last one I had 386.02_1-gnuton1 yesterday. It didn´t save the VLAN ID and priority values but worked well. Even though I risked to upgrade.

I’m not going to change it now again because works fine for me as it is.

Be safe and thanks. It has to be tough to be such a genius man! 😊

Gaizka

Enviado el: viernes, 7 de mayo de 2021 16:08
Para: gnuton/asuswrt-merlin.ng ***@***.***>
CC: gaizkitapowers ***@***.***>; Author ***@***.***>

gaizkitapowers, 2021-05-07:
Hi. I’m honoured, your highness! I understand and looks like I should feel ashamed. Sorry for that.

I’ve downgraded already and it works on 386.2.  It doesn´t keep a stable signal, blinks on the phone now and again but only has to take some ioT devices and the music server. The other 5G works pretty well.

Before the last upgrade yesterday I had 386.02_1-gnuton1. I installed two fans to the router and can´t be sure if there is no involvement in the signal interference. 

I’m not going to change it now again because works fine for me as it is.

Be safe and thanks. It has to be tough to be such a genius man! 😊

Gaizka

gnuton, 2021-05-07:
Yes the fans can be the culprit.
Closing this as invalid.

closed 2021-05-07 by gnuton