gnuton/asuswrt-merlin.ng #103: Build with amas=n possible?

oivano, 2021-05-14:
Hey gnuton , have you tried building with amas disabled recently? I have tried this from your latest dsl-master and that seams impossible. No mater what combination of flags used.  Do you know if disabling amas is at all possible with the latest prebuilds?

build fails with:

ate.o: In function `asus_ate_command':
(.text.asus_ate_command+0xec4): undefined reference to `is_valid_group_id'
nt_mail.o: In function `alert_mail_service':
(.text.alert_mail_service+0x14): undefined reference to `check_wrs_switch'
roamast.o: In function `_assign_roaming_tbl':
(.text._assign_roaming_tbl+0x94): undefined reference to `asusdebuglog'
roamast.o: In function `rast_Proc_INTERNAL_EXAP_RECEIVED':
(.text.rast_Proc_INTERNAL_EXAP_RECEIVED+0x184): undefined reference to `asusdebuglog'
roamast.o: In function `rast_Proc_INTERNAL_EXAP_RECEIVED':
(.text.rast_Proc_INTERNAL_EXAP_RECEIVED+0x238): undefined reference to `asusdebuglog'
roamast.o: In function `rast_Proc_INTERNAL_EXAP_RECEIVED':
(.text.rast_Proc_INTERNAL_EXAP_RECEIVED+0x33c): undefined reference to `asusdebuglog'
roamast.o: In function `_assign_tg_roaming_tbl':
(.text._assign_tg_roaming_tbl+0x8c): undefined reference to `asusdebuglog'
roamast.o:(.text.rast_Proc_INTERNAL_EXAP_BEFORE+0x1ac): more undefined references to `asusdebuglog' follow
collect2: ld returned 1 exit status
make[4]: *** [Makefile:890: rc] Error 1
make[4]: Leaving directory '/home/docker/project/release/src/router/rc'
make[3]: *** [Makefile:3109: rc] Error 2
make[3]: Leaving directory '/home/docker/project/release/src/router'
make[2]: *** [Makefile:608: all] Error 2
make[2]: Leaving directory '/home/docker/project/release/src-rt-6.x.4708'
make[1]: *** [Makefile:5192: bin] Error 2
make[1]: Leaving directory '/home/docker/project/release/src-rt-6.x.4708'
make: *** [Makefile:5373: dsl-ac68u] Error 2

gnuton, 2021-05-14:
Hi,
No i have not tried to disable it, but you MAY need to disable something else to get it building.
I guess that if you have have skilsl, time and motivation you may find an hack-ish way to work around this by linking to the prebuild and then strip the not needed symbols.  That requires a few changes in the Makefiles.. 
 
Not sure what's your target although...

oivano, 2021-05-14:
Everything related to Aimash needs to be also disabled in the target.  This was the last target i used

-DSL-AC68U-NONE := LOGO_LED=y LED_BTN=y ROG=y DWB=y ETHOBD=y FORCE_ROAMING=y BCN_RPT=y BTM_11V=y AMAS=y
+DSL-AC68U-NONE := LOGO_LED=y LED_BTN=y ROG=y

there is probably a dirty way to do it , just wanted to check if you have solved it before, i have seen one of your circlci builds where you have disabled it and your build failed with the same error. Will look at it when i find time.

closed 2021-05-14 by oivano