gnuton/asuswrt-merlin.ng #105: ASUS DSL AX82U Cake not working.

bake-73, 2021-05-28:
**Firmware Version Affected **
384.2_4

**Is this bug present in upstream Merlin releases too?**
Dont have a HND router to try

**Describe the bug**
Cake doesn't work

**To Reproduce**
Steps to reproduce the behavior:
It was nothing special just enabled cake and set wan packet overhead to pppoe vdsl.
It doesn't work at all and it doesn't stick to what i've put in upload and download speeds(it just runs at full speed)
**Expected behavior**
All A's on http://www.dslreports.com/

This was on a fresh install and a reset to defaults before setting up.

Hope this is ok.

gnuton, 2021-05-28:
References (mostly for myself):

- Commit adding Cake QoS 
--- https://github.com/gnuton/asuswrt-merlin.ng/commit/4cb578387ea5d8b3863adbd914daf43c51bbe9d8

- Cake script
--- https://www.snbforums.com/threads/cake-test-cake-experiments-for-386-2-beta.71271/

-- Traffic shaping on linux
--- https://netbeez.net/blog/how-to-use-the-linux-traffic-control/

-- Standard QoS rules per interface on DSL-AX82U
# tc qdisc  show
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth4 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev ptm0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev ppp0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1

Yakumounr, 2021-06-01:
I had tested cake a few days ago in 386.02_4,  finding dslreports showed huge buffer issues I turned traditional back on again and saw in ssh that tc qdisc showed no fq_codel rules only the pfifo_fast , until visiting Advanced_QOSUserRules_Content.asp and hitting apply.
I tried turning QoS off to run a speed test shortly afterwards and then saw the same behaviour with missing fq_codel after re-enabling traditional then, but subsequently I've tried turning QoS off and on again and this behaviour hasn't repeated, perhaps it was only related to cake.

winter2222, 2021-06-10:
Any ETA or idea in which Release we could expect this bug to be fixed ?

gnuton, 2021-06-11:
for DSL-AX82U.. no eta for now,
For sure it won't be part of the next stable which has already a lot of
changes that can give regressions (due to a cleanup).

If you are able to play with scripts I have linked the scripts which
make cake work on the router so you can run it and see what  happens.
I will come back on this ASA i can.

Yakumounr, 2021-06-11:
Thank you for this, I have tried flexqos too and while better than the regular adaptive due to fc_codel, the traditional was still better for my needs. I will give the cake scripts a try.

Yakumounr, 2021-06-11:
I created the two files needed in vi, they look correct to the post linked above, JFFS scripts are enabled, but after restarting qos it does not appear to be working. Dslreports shows bad bufferbloat both upstream and downstream, tc qdisc looks like the default (though one line less than the quote above), and mangle is pretty bare with nothing recognisably cake related in it.

using 386.02_04-gnuton1 with an ax82u

gnuton, 2021-06-13:
I had a look at this and it looks to me that the problem (At least the first one) is related to cake support missing in the kernel of the DSL-AX82U.
In fact sch_cake module is not present in the router 

and 
# tc qdisc add dev br0 root cake 
returns this error:
RTNETLINK answers: No such file or directory

gnuton, 2021-06-13:
cake kernel module is not built for 386.02_6-gnuton0_beta1
The UI seems to not work yet, but setting cake manually seems to be fine now

gnuton, 2021-06-15:
Cake UI doesn't work also in the new Cake v2.
If you wanna test it  run the following command from SSH.
You will see a new cake tab in QoS web page
mkdir -p /jffs/addons/cake-qos && /usr/sbin/curl -s "https://raw.githubusercontent.com/ttgapers/cakeqos-merlin/master/cake-qos.sh" -o "/jffs/addons/cake-qos/cake-qos" && chm
od 755 /jffs/addons/cake-qos/cake-qos && sh /jffs/addons/cake-qos/cake-qos install

Yakumounr, 2021-06-18:
Thank you so much but I can't risk the potential down time from alpha firmware at the moment, I will have to wait and hope someone else will check it is functional.

gnuton, 2021-06-23:
the latest stable has cake support in the kernel. 
The script itself still needs some fixed to work

Yakumounr, 2021-07-09:
For the benefit of others looking to try this -  it still does not currently seem to be possible to get cake running on 386.02_6-gnuton1 or the new  386.03_1-gnuton0_alpha2 at least with the suggestions so far.

cake 1.08 will not install to the dsl-ax82u, the installer states as much by model name, I did not attempt to bypass the check. 
cake 2.0.0 will install the webui which can be viewed but it is not fully functional,  the cake commands in /tmp/qos are never generated by the firmware, cake never starts.

The [experimental script](https://www.snbforums.com/threads/cake-test-cake-experiments-for-386-2-beta.71271/) mentioned earlier in this thread does not solve this as the commands are still missing from /tmp/qos and going by the system log it seems /jffs/scripts/cake-qos is never called.

gnuton, 2021-08-14:
Fixed. 
Cake is now part of the 386.03_2-gnuton0_beta3.

# tc qdisc show
qdisc pfifo_fast 0: dev eth0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth1 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth2 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth3 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth4 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth5 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev eth6 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc pfifo_fast 0: dev ptm0 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc cake 8001: dev ppp0 root refcnt 2 bandwidth unlimited diffserv3 dual-srchost nat nowash no-ack-filter split-gso rtt 100ms noatm overhead 27 
qdisc ingress ffff: dev ppp0 parent ffff:fff1 ---------------- 
qdisc pfifo_fast 0: dev tun21 root refcnt 2 bands 3 priomap 1 2 2 2 1 2 0 0 1 1 1 1 1 1 1 1
qdisc cake 8002: dev ifb4ppp0 root refcnt 2 bandwidth unlimited besteffort dual-dsthost nat wash ingress no-ack-filter split-gso rtt 100ms noatm overhead 27

winter2222, 2021-08-14:
Can confirm it works also for me, thank you very much for this.

closed 2021-08-14 by gnuton