gnuton/asuswrt-merlin.ng #106: Traffic classification missing for upload

gnuton, 2021-05-28:
**Firmware Version Affected **
386.3-alpha1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Activate Adaptive QoS
2. go to classification page
3. see upload is empty

gnuton, 2021-05-29:
to fix we may need to track a different interface:
- eth0 - currently the default option
- ptm0  - tested, not good
- wan0 - under testing.
- eth4 - default on rt-ax58u and possible choice

bake-73, 2021-05-29:
Could this be the problem with cake?

bake-73, 2021-06-26:
Is this fixed in the new version?
Thanks

gnuton, 2021-06-27:
no this is not fixed yet. It's not realated to cake, I do still to get how this really works

gnuton, 2021-08-17:
Tried this again and actually ptm0 worked (-_-) from what I have seen there was not other options.

gnuton, 2021-08-17:
The fix will be part from the next beta 6.

closed 2021-08-17 by gnuton