gnuton/asuswrt-merlin.ng #110: Stability Adjustment section and negative numbers

Hello-olleH, 2021-07-03:
**Firmware Version Affected **
386.02_6-gnuton1

**Describe the bug**
In the original firmware, there are also negative numbers in the Stability Adjustment section, and by selecting these numbers, my internet speed will increase. For example, by selecting the -5, my speed will reach 87294 kbps. There are no negative numbers inside the Merlin firmware and my speed does not exceed 70871 Kbps.

**Screenshots**
[image]
[image]
[image]
[image]
[image]

gnuton, 2021-07-10:
Hi,
This is the new UI which comes with 43xxx GPLs and Merlin is still using 42095.
So it will be updated when we do switch to the new one.

Katana1074, 2021-11-22:
Hi I can confirm that this is fixed in Version 3.0.0.4.386.45660 of the original ASUS firmware, we will just have to be patient and wait for ASUS to release the GPL for 3.0.0.4.386.45660....

Regards

Katana1074

gnuton, 2022-01-05:
This has been fixed. The fix is in the master branch and it will be available along with the next release.

gnuton, 2022-01-05:
this was a duplicate of this issue: https://github.com/gnuton/asuswrt-merlin.ng/issues/136

closed 2022-01-05 by gnuton