gnuton/asuswrt-merlin.ng #111: RT-AX88U only router!

michael1900, 2021-07-19:
Hi. I have this router. Is it possible to have a merlin version for this model router only? I can't believe there is not official merlin version for this model but there is merlin firmware for DSL version. Is it a joke?

gnuton, 2021-07-20:
Hi,
Something is moving... so .please be patient. I will disclose something more on this ASAP.

michael1900, 2021-07-20:
Ok thanks!

targyll, 2021-08-31:
`Something is happening... so .please be patient. I will disclose something more on this ASAP.`

That's good news!
The new TUF Gaming AX5400 has VERY similar specs. Any chance it could be supported in the future?
I bought both models recently and I can't choose which one I should return...

gnuton, 2021-08-31:
Currently I do not have any plan to support AX5400.

bjarchi, 2021-09-01:
@gnuton Any update on AX82U. support? I'm trying to decide whether to keep that hardware or purchase something that already supports merlin.

gnuton, 2021-09-01:
I have plans to support it. i have got one unit and i am just waiting for
Asus to deliver the new GPL.

bjarchi, 2021-09-01:
Fun, so that might be days or months? Any sense of how this hardware compares to the older higher end hardware like 88u?

gnuton, 2021-09-01:
hopefully we will get these GPLs in days.
As for comparison hw comparison there is already a lot of to read on the
forum. Please have a look there.

bjarchi, 2021-09-01:
Do you have a link to the forum? I don't know which one people tend to use; all I've seen doc wise is the merlin wiki.

alliedcvil, 2021-09-04:
@gnuton Thank you so much for this it will be great to have that added!

gnuton, 2021-09-10:
unfortunately it looks like Asus is taking a little bit of time to release
the GPLs again.
https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/RT-AX82U/HelpDesk_BIOS/
as you can see there are no GPL (source code) available for this router yet
at this moment.

targyll, 2021-09-25:
Why are they waiting for so long???

gnuton, 2021-09-26:
I have contacted Asus, they promised me there is just 1-2 weeks to wait.
Once out there will be also a new Merlin version.
Asus devs are fixing stuff in the GPLs, it's just one time fix. So the next
GPLs won't be so slow to come out anymore.. the waiting is almost over! :D

gnuton, 2021-09-29:
Good news. We got the first GPLs from ASUS.
I ahve not got yet the RT-AX82U but it's now matters of days!

kurokawaikki, 2021-10-16:
Thank you for the great work!
I just got rt-ax82u yesterday. I can't wait to flash merlin onto it.

anarchy9388, 2021-10-17:
Any updates here? 👍

gnuton, 2021-10-17:
I have not received the GPL source code from Asus for this model yet, but two days ago they told me it s ready and under verification. I cannot do anything if they do not send me the GPL. My expectation is to have something in two weeks. I apologize for delay.

gnuton, 2021-11-01:
Hi there,
Today I built  the first Merlin RT-AX82U image.
I will test it this evening when back home. 
I will be sharing it for a brodader testing soon.
Stay tuned! :D

anarchy9388, 2021-11-01:
Thanks @gnuton !

gnuton, 2021-11-01:
First alpha image is now available.

[image]

anarchy9388, 2021-11-01:
Thanks @gnuton , I'll give it a whirl!

AlexRichter, 2021-11-01:
I gave this a try and it seems that whenever I inputted my internet details in setup it would cause the router to go into a crash loop. 

Using it in access point mode worked though.

AlexRichter, 2021-11-02:
Seems my issue was related to this PPPoE issue: https://www.snbforums.com/threads/386-4-alpha-build-s-testing-available-build-s.74186/page-13#post-722191

gnuton, 2021-11-02:
@AlexRichter yep.
Eric sent me also the patches to fix that.
Just for reference for my self.. so that I do not forget them.. here they are:

1. https://github.com/RMerl/asuswrt-merlin.ng/commit/e3ee6f40792136cd4f591c423a3242727b6ba13c
2. https://github.com/RMerl/asuswrt-merlin.ng/commit/23082e245288629aa3caeca1adcff213e3c6c485

anarchy9388, 2021-11-02:
Yeah mines in the crash loop also... how do I get it out of the crash loop?

AlexRichter, 2021-11-02:
@anarchy9388 turn the router off, hold the WPS button, turn the router back on, release the WPS button when the power light is completely off, should boot normally then

anarchy9388, 2021-11-02:
thanks @AlexRichter

anarchy9388, 2021-11-02:
@gnuton https://github.com/gnuton/asuswrt-merlin.ng/pull/126

gnuton, 2021-11-02:
Thanks @anarchy9388 but I have already merged locally the upstream changes and it comes with the fix in your PR.... the problem is that it doesn't build.. (not an issue of your PR) :D  and I do not have any clue as for now of what's wrong now :D 
As for the tmctl patch I am not sure it's needed yet. from the build logs it may seems to be fine.
Beside that I fixed entware.... 
I will publish the new image once it builds again.

anarchy9388, 2021-11-02:
sounds good , yeah Idk what is going on with the build either haha. not familiar enough with the code

gnuton, 2021-11-03:
Here is the new image.
Things to  test in this image:
- amtm can install entware packages
- crash loop in PPPoE
- AiProtection/Trend Micro enabled 

https://1511-175516287-gh.circle-artifacts.com/0/workspace/release/RT-AX82U_386.4_alpha2_cferom_puresqubi.w

AlexRichter, 2021-11-03:
The PPPoE no longer crash loops for me but NTP wasn't working no matter what I tried, which prevented testing amtm.
Reverting to stock firmware NTP is working fine.

gnuton, 2021-11-03:
Thanks for testing Alex.
there is still a lot of things to be fixed.. I will check and keep you
updated.
If you guys have any preference on the feature you wanna get fixed please
let me know.
Otherwise I will pick things randomly...

tconnard, 2021-11-03:
Ntp is an issue for me as well. 
I manually set the time and set ntp_ready=1
Amtm still failed to install entware, I think it expects /opt to be a link to /temp/opt, but it is a directory with files in it on the squashfs.
A bind mount may also work.

gnuton, 2021-11-03:
This fixes NTP
https://1518-175516287-gh.circle-artifacts.com/0/workspace/release/RT-AX82U_386.4_alpha2_cferom_puresqubi.w

keenjeen, 2021-11-06:
Thanks for the hardwork! I've installed the latest version in my RT-AX82U and start to try on the features. These are some of the problems that I observed:

1. For 5GHz wireless network, the channel bandwidth seems to only works in 20MHz irregardless of the bandwidth setting. I have couple of AX device which can work with 80/160 MHz but it's only connected at 20MHz bandwidth.

2. The WebGui only show connected clients via ethernet cables but not wifi connected clients. Wifi clients can only be seen under the wireless log page

3. Wifi radar tab showing 404 Not found

Once again, thank you very much and I'm glad that my router can finally works with Merlin :)

keenjeen, 2021-11-06:
The 'Apply' button under 'Wireless - General' page unable to apply the new changes to the system. Did a factory restore and it fixed the problem but after setting some some other parameters, the 'Apply' button doesn't work again.

This is what I got from the developer console
[image]

washran, 2021-11-07:
I have the same problems as keenjeen. I installed the firmware on my RT-AX82U and I also noticed in the system log there's a lot of Nov  6 23:02:43 rc_service: watchdog 1237:notify_rc start_cfgsync lines. GNUton, thanks for your hard work.

gnuton, 2021-11-07:
Hi,
Thanks for reporting this.
Collecting the issues here and marking here.

- [ ] For 5GHz wireless network, the channel bandwidth seems to only works in 20MHz irregardless of the bandwidth setting. I have couple of AX device which can work with 80/160 MHz but it's only connected at 20MHz bandwidth.

- [ ]  The WebGui only show connected clients via ethernet cables but not wifi connected clients. Wifi clients can only be seen under the wireless log page

- [ ] Wifi radar tab showing 404 Not found

- [ ] AiMesh is empty

- [ ] The 'Apply' button under 'Wireless - General' page unable to apply the new changes to the system.

gnuton, 2021-11-07:
@keenjeen 
Analysis of that issue: 
  A. var current_node_count = [][0].length - 1;
  B the code that generate that is "var current_node_count = [<% get_cfg_clientlist(); %>][0].length - 1;"
  C so the problem is that get_cfg_clientlist somehow doesn't return anything... that's a web hook

keenjeen, 2021-11-07:
> @keenjeen
> Analysis of that issue:
> A. var current_node_count = [][0].length - 1;
> B the code that generate that is "var current_node_count = [<% get_cfg_clientlist(); %>][0].length - 1;"
> C so the problem is that get_cfg_clientlist somehow doesn't return anything... that's a web hook

@gnuton 
I assume this is caused by connected wireless clients is not being displayed. I get similar error when I clicked on the "View List' under network map. Please let me know if there is any information that I can help to provide to you.

[image]

gnuton, 2021-11-07:
@keenjeen 
This image has several fixes and not it's using the correct Advanced wirless page for RT-AX82U
After resetting the router to factory I cannot reproduce issue I5 and in the console I get always get_cfg_clientlist.
There is still something wrong.. but maybe this is a step forward...
https://1522-175516287-gh.circle-artifacts.com/0/workspace/release/RT-AX82U_386.4_alpha2_cferom_puresqubi.w

keenjeen, 2021-11-07:
@gnuton 
I'll try now and let you know in a while

keenjeen, 2021-11-07:
@gnuton 

I've updated the firmware and able to reproduce the issue in this sequence
1. Factory default
2. Quick Setup(2 separate SSID for 2.4 and 5GHz network)
3. Enable DFS channel(Apply button still working at this stage)
4. Reboot router using 'Reboot' button on top of the webgui

After the reboot, the "Apply" button no longer works and console is showing the same error with the get_cfg_clientlist

Thanks!

gnuton, 2021-11-07:
Confirmed DFS breaks something in the shared mem.. I will keep on trygin to understand what's the problem.. thanks for testing @keenjeen 
As for Issue I7 the problem related to these themes is that they require ROG to be enabled.

keenjeen, 2021-11-07:
@gnuton 
I did another test just to confirm whether the DFS is the root cause for this but it's not
1. Factory restore
2. Quick Setup
3. Check the 'Apply" button(without changing any setting, there is a pop up for 'Applying settings', console not showing any error)
4. Reboot using the webgui
5. Check the 'Apply' button again

After the reboot, the 'Apply' button start to stop functioning. Seems like the reboot is causing I5

keenjeen, 2021-11-07:
@gnuton 
Did another test to reboot using power button instead of the 'Reboot' button. the 'Apply' button stop functioning after the router restarted.

washran, 2021-11-07:
@gnuton 

You got the 2.4 ghz wireless to show 20/40 mhz and 5 ghz to show 20/40/80/160 mhz. Both 2.4 and 5 show 20 mhz in system wireless log. Apply buttons not working correctly. Wireless apply button still not functioning on webgui. In all, I'm having similar issues as @keenjeen ...

hegoku, 2021-11-11:
I used your docker image to build RT-AX82U, but errors happened.

>   CC [M]  net/sched/em_u32.o
  CC [M]  net/sched/em_meta.o
  CC      net/ipv4/netfilter/iptable_nat.o
  CC      net/netfilter/xt_CLASSIFY.o
  CC      net/ipv4/netfilter/iptable_raw.o
make[7]: *** No rule to make target 'net/netfilter/xt_DSCP.o', needed by 'net/netfilter/built-in.o'.  Stop.
make[7]: *** Waiting for unfinished jobs....
  CC      net/netfilter/xt_CT.o
  CC [M]  net/ipv4/ip_gre.o
  CC [M]  net/ipv4/ah4.o
  CC [M]  net/sched/em_text.o
  CC      net/ipv4/netfilter/ipt_MASQUERADE.o
  CC      net/ipv4/netfilter/ipt_REJECT.o
  CC [M]  net/ipv4/esp4.o
  CC [M]  net/ipv4/ipcomp.o
make[6]: *** [scripts/Makefile.build:496: net/netfilter] Error 2
make[6]: *** Waiting for unfinished jobs....
  CC [M]  net/ipv4/xfrm4_tunnel.o
  CC [M]  net/sched/em_ipset.o
  CC      net/ipv4/netfilter/ipt_TRIGGER.o
  CC [M]  net/ipv4/tcp_bic.o
  CC [M]  net/ipv4/tcp_westwood.o
  CC [M]  net/ipv4/tcp_htcp.o
  CC      net/ipv4/netfilter/dnsmq.o
  CC [M]  net/ipv4/netfilter/nf_nat_h323.o
  CC [M]  net/ipv4/netfilter/nf_nat_pt.o
  LD      net/sched/sch_cake/built-in.o
  CC [M]  net/sched/sch_cake/sch_cake.o
  CC [M]  net/ipv4/netfilter/nf_nat_rtsp.o
  LD      net/ipv4/gre.o
  CC [M]  net/ipv4/netfilter/ipt_tos.o
  CC [M]  net/ipv4/netfilter/ipt_ipp2p.o
  CC [M]  net/ipv4/netfilter/ipt_web.o
  CC [M]  net/ipv4/netfilter/ipt_webmon.o
  CC [M]  net/ipv4/netfilter/ipt_TOS.o
  LD      net/ipv4/netfilter/nf_conntrack_ipv4.o
  LD      net/ipv4/netfilter/nf_nat_ipv4.o
  LD      net/ipv4/netfilter/built-in.o
  LD      net/sched/built-in.o
  LD      net/ipv4/built-in.o
make[5]: *** [Makefile:1254: net] Error 2
make[5]: Leaving directory '/build/release/src-rt-5.02axhnd.675x/kernel/linux-4.1'
make[4]: *** [Bcmkernel.mk:107: default] Error 2
make[4]: Leaving directory '/build/release/src-rt-5.02axhnd.675x/build'
make[3]: *** [build/Makefile:428: kernelbuild] Error 2
make[3]: Leaving directory '/build/release/src-rt-5.02axhnd.675x'
make[2]: *** [Makefile:609: all] Error 2
make[2]: Leaving directory '/build/release/src-rt-5.02axhnd.675x'
make[1]: *** [Makefile:5188: bin] Error 2
make[1]: Leaving directory '/build/release/src-rt-5.02axhnd.675x'
make: *** [Makefile:5376: rt-ax82u] Error 2

hegoku, 2021-11-11:
> *** No rule to make target 'net/netfilter/xt_DSCP.o', needed by 'net/netfilter/built-in.o'. Stop.

I find the reason is case sensitive. I rename `'net/netfilter/xt_dscp.o` to `'net/netfilter/xt_DSCP.o` which solves the problem.

And these files have the same isses:

`'net/netfilter/xt_tcpmss.o` to `net/netfilter/xt_TCPMSS.o`

gnuton, 2021-11-11:
Hey Jack,
Are you running docker in windows?
you should not need to rename built files.

hegoku, 2021-11-11:
> Hey Jack, Are you running docker in windows? you should not need to rename built files.
> […](#)

No, I use macos.

gnuton, 2021-11-12:
here the log of my last build btw, if youare interested.
It looks like make generates both in my case:
https://circleci.com/api/v1.1/project/github/gnuton/asuswrt-merlin.ng/1525/output/104/0?file=true&allocation-id=618858d986ce8e417b61a41f-0-build%2F1B3EE45

hegoku, 2021-11-12:
> here the log of my last build btw, if youare interested. It looks like make generates both in my case: https://circleci.com/api/v1.1/project/github/gnuton/asuswrt-merlin.ng/1525/output/104/0?file=true&allocation-id=618858d986ce8e417b61a41f-0-build%2F1B3EE45
> […](#)

Ok. I will take a look.

kxndal, 2021-11-12:
Any new Build to test?

gnuton, 2021-11-12:
https://1526-175516287-gh.circle-artifacts.com/0/workspace/release/RT-AX82U_386.4_alpha2_cferom_puresqubi.w
Enabled AHS. 
Previous bugs are still there althogh..

gilvbp, 2021-11-12:
@gnuton  hi there,  does aimesh function it's working good?

gnuton, 2021-11-12:
Nope. There is a lot to fix. Please check the check list above to know what
a broken

washran, 2021-11-12:
> https://1526-175516287-gh.circle-artifacts.com/0/workspace/release/RT-AX82U_386.4_alpha2_cferom_puresqubi.w Enabled AHS and a few other things. but the previous bugs are still there althogh..

I installed the newest firmware. The Wifi Radar is still 404 not found. The 2.4 and 5 wifi is still locked at 20 mhz according to system log wireless log. Some apply buttons work, some don't. I find it weird this rt-ax82u router is the weirdest one since the rt-ax58u and rt-ax3000 use the same hardware but the firmware is another story. I used to have the rt-ax58u and had no problems with asuswrt-merlin... @gnuton I appreciate your hard work.

gilvbp, 2021-11-12:
> Nope. There is a lot to fix. Please check the check list above to know what a broken
> […](#)

How can I help? Do you guys are using irc or slack or something to discuss the fixes?

[28 more replies]

closed 2021-11-17 by gnuton