gnuton/asuswrt-merlin.ng #112: No icons in webgui when using IP through HTTP

maicol07, 2021-07-24:
**Firmware Version Affected **
386.02_6

**Is this bug present in upstream Merlin releases too?**
I don't know

**Describe the bug**
Can't see icons when accessing the modem (DSL-AX82U) through http://192.168.1.1.
Works when using HTTPS and/or `router.asus.com`

**To Reproduce**
Steps to reproduce the behavior:
1. Open http://192.168.1.1 (I have used Firefox)
2. Go to WAN --> Port forwarding or WAN --> Internet connection
3. There are no icons (check table in screenshot)

**Expected behavior**
Icons showing

**Screenshots**
[image]

gnuton, 2021-07-26:
mmm it feels like an issue with your browser.
could you please check to run it in incognito? Does it behave the same?

Another test you can do is to open the chrome tools (by presssing F12 if you don't have a PC) and check the requests.
If the requests fail than the bug is valid.

maicol07, 2021-07-26:
Now works. I think upgrading to alpha4 fixed the issue

closed 2021-07-26 by maicol07