gnuton/asuswrt-merlin.ng #113: DSL-AC68U - System Log filled with DSL: Upload DSL firmware failed

mrptsai, 2021-08-01:
Hi!

Current Version : 386.02_6-gnuton1

I obviously don't use DSL on my DSL-AC68U with your firmware, but does the System Log need to fill up with the following?

```
Aug  1 03:01:13 DSL: Upload DSL firmware failed
Aug  1 03:01:23 DSL: Try to upload DSL firmware
Aug  1 03:01:36 DSL: Upload DSL firmware failed
Aug  1 03:01:46 DSL: Try to upload DSL firmware
Aug  1 03:01:59 DSL: Upload DSL firmware failed
Aug  1 03:02:09 DSL: Try to upload DSL firmware
Aug  1 03:02:22 DSL: Upload DSL firmware failed
Aug  1 03:02:32 DSL: Try to upload DSL firmware
Aug  1 03:02:45 DSL: Upload DSL firmware failed
Aug  1 03:02:55 DSL: Try to upload DSL firmware
Aug  1 03:03:08 DSL: Upload DSL firmware failed
Aug  1 03:03:18 DSL: Try to upload DSL firmware
Aug  1 03:03:31 DSL: Upload DSL firmware failed
Aug  1 03:03:41 DSL: Try to upload DSL firmware
Aug  1 03:03:54 DSL: Upload DSL firmware failed
Aug  1 03:04:04 DSL: Try to upload DSL firmware
Aug  1 03:04:17 DSL: Upload DSL firmware failed
Aug  1 03:04:27 DSL: Try to upload DSL firmware
Aug  1 03:04:40 DSL: Upload DSL firmware failed
Aug  1 03:04:50 DSL: Try to upload DSL firmware
Aug  1 03:05:02 DSL: Upload DSL firmware failed
``` 

Thanks

gnuton, 2021-08-02:
Could you please add the step to reproduce this?

mrptsai, 2021-08-02:
I installed 386.02_6-gnuton1 through the Administration Menu on the Firmware Tab after downloading it from this github repo and using the Manual firmware update

gnuton, 2021-08-02:
More than a bug it looks to me that there is something wrong with:
A. the download of the image, maybe the file you are flashing is corrupted
B there is an hardware failure on your side
What that message tells me is that  the flashing of the DSL modem is failing.

Do  you get the same message if you flash the original firmware?
Is the DSL modem working?

mrptsai, 2021-08-02:
I will try the original firmware, check the log and and then try reinstalling your awesome firmware.

Thank you.

mrptsai, 2021-08-02:
No hardware failure that I am aware of

gnuton, 2021-08-06:
I will close this since for me it works. 
Again this may be something related to an hardware issue or bad downloaded image.

closed 2021-08-06 by gnuton

mrptsai, 2021-08-06:
Yes. No problem. Thanks. I’ll let you know what I find.

gnuton, 2021-08-06:
Feel free to update this thread with yuour findings I will try to help you as much as I can and try to understand what's the issue.
But as for now I do not see this coming from the image that I released.

If needed we can re-open this without any problem.

site-speed, 2022-07-14:
I've recently had this issue running on the latest ASUS firmware, and still saw it with your latest firmware, but then it seemed to fix it.

Best way to reproduce it (but not all the time) seems to be allow and automatic update of firmware.
trigger a power cut - cut power to the ASUS power supply block not just the switch on the back of the unit.
Power back up.

Details below - sorry for the long post but hopefully something in there might be of help.

About a month ago, doing some DIY I tripped the main home circuit breaker, shutting of power to all sockets.
After flipping it back on noticed after a while I had no internet.
ASUS DSL-AC68U was up and running, Wi-Fi working, internal LAN connections working but no DSL or Internet lights showing on router.
Rebooted it (power off/on with power switch) - same thing
Home page of GUI said that the "phone cable was unplugged" - which it wasn't

Noticed in the log the messages:

May  5 05:07:01 DSL: Try to upload DSL firmware
May  5 05:07:14 DSL: Upload DSL firmware failed
May  5 05:07:24 DSL: Try to upload DSL firmware
May  5 05:07:37 DSL: Upload DSL firmware failed

Looking on the firmware page on router GUI could indeed see that there were no details displayed in the DSL firmware section

Noticed that "Auto upgrade firmware" option was switched on (IIRC I'd not seen it before and IIRC had not switched it on but could be wrong)
Firmware version installed on router was one higher than the files I'd downloaded
	- I always download the rescue and full files and then upgrade via the manual upload option
Suggested to me that the auto update had happened maybe a month or two before, something didn't quite work as expected and after the power loss router was in a sad state.

Tried uploading firmware (same newer version)
Tried uploading firmware (previous version that I had saved away)

No luck - oh dear!  Was thinking maybe power cut had fried something

Tried various random things (all with no joy)
Replace phone cable with another
Checked a cable was working by digging out ISP provided modem plugging that in and firing it up
	- DSL light green after just a few seconds
Tried reloading full firmware via GUI
Read about "Recovery mode" on ASUS support pages - tried that with recovery firmware a few times using their recovery tool
Tried factory reset and then more of the above
At some point I must have carried out the "magic" sequence as it sprang back into life
	- Proving that modem wasn't fried by power cut at least
DSL driver version showing on the firmware page (but I now later realise that firmware version wasn't)
However, after the various factory resets had lost all my config (e.g. run on 10.x.x.x, various IPs reserved in DHCP, etc.)
Put back on previous version of router firmware
Reloaded my saved config from that version - thinking all would now be fine.
BOOM! Back to square one - no internet connection, no DSL firmware version info showing
Random sequenced firmware uploads, factory resets didn't bring it back to life
	- Though I knew now it *should* be possible

Next day decided to Google the "DSL: Upload DSL firmware failed" message
	- Found a number of posts describing the same issue
	- Common themes were:
	a) Issue happened after an automatic firmware update
	b) Issue happened after a manual update to a particular (ancient) firmware release and a later reboot or power cycle
	c) Issue happened after a power cut of some sort

Links here:
[DSL-AC68u missing DSL firmware - Asus (whirlpool.net.au)](https://forums.whirlpool.net.au/archive/36pxm5p3)

[Replacing Asus DSL-AC68U - recommendations - Modems/Routers (whirlpool.net.au)](https://forums.whirlpool.net.au/archive/3484nq73)
[Modem Issue on DSL-AC68U - Asus (whirlpool.net.au)](https://forums.whirlpool.net.au/archive/322y5073)

[DSL-AC68U 3.0.0.4.384_21126 bug ((( FIXED ))) | SmallNetBuilder Forums (snbforums.com)](https://www.snbforums.com/threads/dsl-ac68u-3-0-0-4-384_21126-bug-fixed.47573/)

An ASUS representative chimed in on some of these threads (the top one above) and suggested the following sequence:
	1) Factory reset
	2) Upload recovery firmware using recovery tool (I guess it’s a TFTP client)
	3) Upload via GUI full firmware
	4) Upload via GUI full firmware a second time
[... 56 lines cut ...]
BOOM! No DSL light again!

Time for bed

Next day tried ASUS suggest magic sequence of updating firmware (no factory reset tho)
No luck
Tried enabling Dual WAN - when running on Merlin this seemed to fix things
No luck

Decided to start again the process with Merin firmware
Tried uploading via GUI, it went through the process but once GUI back again still showing ASUS firmware
Repeated - same again

Put router in recovery mode
Uploaded latest Merlin via CFE
	- IIRC it was the version *without* DSL firmware (my plan was to start from scratch)

Once it restarted, DSL light back on, internet connection back, all my config in place
Both DSL firmware version and DSL driver version sowing on the GUI.

Not touching it ever again!

Possibly coincidence, but possibly related.
Both times before the issue was fixed I had temporarily plugged in and fired up my ISP provided modem.
Possibly (but unlikely perhaps) that this somehow reset something on the line or at DSLAM end, which then enabled the ASUS to properly re-instate the DSL firmware.

site-speed, 2022-07-24:
> @site-speed<https://github.com/site-speed> I cannot see any possible relation between DSLAM end and the firmware itself
> The DSL-AX82U is made of two distinct units, the router and the DSL modem.
> if you have upload DSL firmware failed you must probably have a faulty hardware DSL modem.

Thanks Antonio for the speed reply
– sorry for the delay getting back to you.

In reverse order:

  *   if you have upload DSL firmware failed you must probably have a faulty hardware DSL modem.

Yes I’m not ruling this out – though:

  *   On both occasions - once I managed to get the DSL firmware to seemingly re-upload to the DSL module – it then worked just fine.
     *   Without wanting to jinx it mine has now been running perfectly well for the last 6-8 weeks.
  *   On the forum posts I found, some people reported having this same issue with a brand new out of the box DSL-AC68U   (not sure if it affects any other models)
  *   At least one official ASUS firmware release triggered this problem for multiple users
So I believe there’s a chance that the hardware (and/or in combination with user adjusted config) can get into a state where the DSL firmware fails to upload to the DSL module

If I get the chance I’ll have search in the code for the specific error messages and look in their vicinity for what may cause/trigger them.

  *   The DSL-AX82U is made of two distinct units, the router and the DSL modem.

  *   I have the DSL-AC68U; but presumably this also has separate units.
  *   As mentioned above I don’t know if the issue in question happens on devices other than the DSL-AC68U

  *   I cannot see any possible relation between DSLAM end and the firmware itself

  *   I’d be surprised as well; unless unplugging the cable, connecting a working modem for a while, then reconnecting the problematic one can somehow reset some state ant one end or the other.
  *   Just wanted to flag it for completeness, in case it helps someone.
     *   I’ll chalk it down as coincidence that shortly after doing the above, on two separate occasions, the issue got resolved.
     *   Will look out for it though if for some reason I find myself in the same scenario in the future.

Thanks again,

Tim