gnuton/asuswrt-merlin.ng #115: Wifi ping focus

MHKalantarian, 2021-08-16:
**Firmware Version Affected **
386.02_6
**Is this bug present in upstream Merlin releases too?**
IDK

**Describe the bug**
This is a very confusing problem, could be a feature that I can disable/enable but I don't know about it, router focuses ping only on used device, for example if I'm only connected to wifi with my phone, ping between phone and router is 50~150ms but if I start using continues data on it, like video call in whatsapp the ping will lock on 1ms! It happens on all my wireless devices

**To Reproduce**
1.Connect a device to 5ghz wifi without any configs (newly factory reset). 
2.Connect any wireless device and ping it from pc, or router. for me it's usually high or semi-low with spikes
3.Start using continuous data with the wireless device (for example video call or watch youtube)
4.Watch the ping graph/cli the ping locks on lowest it can get 1~2ms

**Expected behavior**
Ping should always stay 1~2ms cause I wanna stream video between my devices with high bitrate and I haven't been able to because of this problem. (yes i'm sure thats the problem!)

**Screenshots**
Top device is idle, it also stays the same if I start streaming video to it
Bottom device is using internet (whatsapp videocall)
[image]

gnuton, 2021-08-17:
Hi, 
Please open a thread in the forum and link it here. This case has to be understood more carefully and there are more people who can say why this behaviour happens eg: Wifi power management of your devices (not the router).

gnuton, 2021-09-20:
I will close this for now,, 
for me as described is not a bug at all. 
When device is idle -> it goes in power saving and ping gets higher to save energy
When device is in a real time stream  as communication traffic -> ping goes lower
When device is streaming media -> usually you do not get a continuos stream but bursts/caching cicles. This traffic is by default not prioritized and goes through HTTP and not UDP datagrams as for the communications one...

All these behaviours may change using QoS.

closed 2021-09-20 by gnuton